chipsalliance/caliptra-sw #851: Verilator test failures: caliptra-runtime::integration_tests

korran, 2023-09-20:
```
50364:2023-09-20T01:45:54.8524368Z         FAIL [26955.341s] caliptra-runtime::integration_tests test_disable_attestation_cmd
52108:2023-09-20T02:35:19.0550460Z         FAIL [27705.854s] caliptra-runtime::integration_tests test_fips_cmd_api
53341:2023-09-20T03:14:50.3249292Z         FAIL [27293.130s] caliptra-runtime::integration_tests test_invoke_dpe_sign_and_certify_key_cmds
54099:2023-09-20T03:42:33.9457981Z         FAIL [27095.727s] caliptra-runtime::integration_tests test_keyvault
56727:2023-09-20T05:58:25.2853509Z         FAIL [38798.178s] caliptra-runtime::integration_tests test_fw_info
57214:2023-09-20T06:15:38.4889439Z         FAIL [19932.771s] caliptra-runtime::integration_tests test_stash_measurement
```
```
2023-09-20T06:15:38.4889439Z         FAIL [19932.771s] caliptra-runtime::integration_tests test_stash_measurement
2023-09-20T06:15:38.4889986Z 
2023-09-20T06:15:38.4890783Z --- STDOUT:              caliptra-runtime::integration_tests test_stash_measurement ---
2023-09-20T06:15:38.4891261Z 
2023-09-20T06:15:38.4891437Z running 1 test
2023-09-20T06:15:38.4895722Z       6,155 ready_for_fuses is high
2023-09-20T06:15:38.4896273Z       6,333 writing to cptra_bootfsm_go
2023-09-20T06:15:38.4896932Z test test_stash_measurement has been running for over 60 seconds
2023-09-20T06:15:38.4897533Z     195,201 UART: 
2023-09-20T06:15:38.4902426Z     195,256 UART: Running Caliptra ROM ...
2023-09-20T06:15:38.4902789Z     196,662 UART: 
2023-09-20T06:15:38.4903077Z     202,819 UART: [state] CFI Enabled
2023-09-20T06:15:38.4903441Z     205,066 UART: [state] LifecycleState = Unprovisioned
2023-09-20T06:15:38.4903787Z     207,326 UART: [state] DebugLocked = No
2023-09-20T06:15:38.4904235Z     209,296 UART: [state] Watchdog Timer is not started because the device is not locked for debugging
2023-09-20T06:15:38.4922829Z     355,872 UART: ROM Digest: BBB6BA0BFBA2F1E2103A1060EB333FF6613A1F43E352AA823F62EF22650FF80F
2023-09-20T06:15:38.4923719Z     367,112 UART: [kat] ++
2023-09-20T06:15:38.4924174Z     367,624 UART: [kat] sha1
2023-09-20T06:15:38.4924963Z     376,784 UART: [kat] SHA2-256
2023-09-20T06:15:38.4925523Z     379,656 UART: [kat] SHA2-384
2023-09-20T06:15:38.4926130Z     383,921 UART: [kat] SHA2-384-ACC
2023-09-20T06:15:38.4926750Z     386,218 UART: [kat] ECC-384
2023-09-20T06:15:38.4927341Z   3,956,572 UART: [kat] HMAC-384
2023-09-20T06:15:38.4927802Z   3,967,624 UART: [kat] LMS
2023-09-20T06:15:38.4928827Z   5,069,426 UART: [kat] --
2023-09-20T06:15:38.4929416Z   5,073,687 UART: [cold-reset] ++
2023-09-20T06:15:38.4929925Z   5,075,903 UART: [idev] ++
2023-09-20T06:15:38.4930397Z   5,076,482 UART: [idev] CDI.KEYID = 6
2023-09-20T06:15:38.4930925Z   5,077,732 UART: [idev] SUBJECT.KEYID = 7
2023-09-20T06:15:38.4931615Z   5,079,204 UART: [idev] UDS.KEYID = 0
2023-09-20T06:15:38.4932156Z   5,098,333 UART: [idev] Erasing UDS.KEYID = 0
2023-09-20T06:15:38.4933038Z   8,365,543 UART: [idev] Using Sha1 for KeyId Algorithm
2023-09-20T06:15:38.4933672Z   8,387,845 UART: [idev] --
2023-09-20T06:15:38.4934112Z   8,393,994 UART: [ldev] ++
2023-09-20T06:15:38.4934621Z   8,394,587 UART: [ldev] CDI.KEYID = 6
2023-09-20T06:15:38.4939743Z   8,395,829 UART: [ldev] SUBJECT.KEYID = 5
2023-09-20T06:15:38.4940690Z   8,397,301 UART: [ldev] AUTHORITY.KEYID = 7
2023-09-20T06:15:38.4941196Z   8,398,883 UART: [ldev] FE.KEYID = 1
2023-09-20T06:15:38.4941550Z   8,412,000 UART: [ldev] Erasing FE.KEYID = 1
2023-09-20T06:15:38.4942210Z  11,700,057 UART: [ldev] Signing Cert with AUTHORITY.KEYID = 7
2023-09-20T06:15:38.4942800Z  15,298,320 UART: [ldev] PUB.X = 504D38CA45D997901F48BA333A149A2FB2668B973AAC64D3B79ECC09A663F02ED2FABA133F5FA499677AEB7687C99B25
2023-09-20T06:15:38.4943542Z  15,314,018 UART: [ldev] PUB.Y = A5CD31125698B3322C086E4398A591946BF20A3ECEA8EB7D7C23410D49FA877E7ECFBE47BD883BD53D7DA865AC217F1D
2023-09-20T06:15:38.4944545Z  15,330,088 UART: [ldev] SIG.R = 14B5D494DF875AC7CEE9FE3E9A6168A6B64BFB01171F5ECC5216C8BB945FAF957CED1BE6C7CFBB1DDB2B7F4BFD76A492
2023-09-20T06:15:38.4945354Z  15,346,000 UART: [ldev] SIG.S = 821D8CC92F40AFF934771D7652C9BC3C59D47E5635E110A8486B76F7433B02B3C6B0FD8B1B6548B5994C481547111FC5
2023-09-20T06:15:38.4945987Z  15,367,949 UART: [ldev] --
2023-09-20T06:15:38.4946282Z  15,375,413 ready_for_fw is high
2023-09-20T06:15:38.4946626Z  15,375,415 <<< Executing mbox cmd 0x46574c44 (63908 bytes) from SoC
2023-09-20T06:15:38.4947021Z  15,375,474 UART: [fwproc] Waiting for Commands...
2023-09-20T06:15:38.4959765Z  15,411,190 UART: [fwproc] Received command 0x46574c44
2023-09-20T06:15:38.4960559Z  15,413,803 UART: [fwproc] Received Image of size 63908 bytes
2023-09-20T06:15:38.4961332Z  18,131,031 UART: [fwproc] Image verified using Vendor ECC Key Index 0
2023-09-20T06:15:38.4962176Z  18,214,988 UART: [fwproc] Loading FMC at address 0x40000000 len 12964
2023-09-20T06:15:38.4962866Z  18,296,394 UART: [fwproc] Loading Runtime at address 0x40004000 len 45060
2023-09-20T06:15:38.4963708Z  18,572,382 >>> mbox cmd response: success
2023-09-20T06:15:38.4964675Z  18,575,639 UART: [afmc] ++
2023-09-20T06:15:38.4965373Z  18,576,219 UART: [afmc] CDI.KEYID = 6
2023-09-20T06:15:38.4966175Z  18,577,469 UART: [afmc] SUBJECT.KEYID = 7
2023-09-20T06:15:38.4966936Z  18,578,927 UART: [afmc] AUTHORITY.KEYID = 5
2023-09-20T06:15:38.4967384Z  21,893,593 UART: [afmc] Signing Cert with AUTHORITY.KEYID = 5
2023-09-20T06:15:38.4967814Z  24,186,864 UART: [afmc] Erasing AUTHORITY.KEYID = 5
2023-09-20T06:15:38.4968432Z  25,495,691 UART: [afmc] PUB.X = 74A2DA8F43E3CF7B034EB91F9FEBBBEBD3F2BB8B9B15CCF6B2944AE9C548D4ACDD20742598EA46075F5EE3D570A3C884
[... 36 lines cut ...]
2023-09-20T06:15:38.4984872Z  32,570,643 UART:  / ___|__ _| (_)_ __ | |_ _ __ __ _  |  _ \_   _|
2023-09-20T06:15:38.4985375Z  32,572,309 UART: | |   / _` | | | '_ \| __| '__/ _` | | |_) || |
2023-09-20T06:15:38.4985787Z  32,573,908 UART: | |__| (_| | | | |_) | |_| | | (_| | |  _ < | |
2023-09-20T06:15:38.4986115Z  32,575,507 UART:  \____\__,_|_|_| .__/ \__|_|  \__,_| |_| \_\|_|
2023-09-20T06:15:38.4986422Z  32,577,106 UART:                |_|
2023-09-20T06:15:38.4986687Z  32,577,748 UART: 
2023-09-20T06:15:38.4987006Z  32,674,348 UART: Caliptra RT listening for mailbox commands...
2023-09-20T06:15:38.4987414Z  32,675,908 <<< Executing mbox cmd 0x4d454153 (108 bytes) from SoC
2023-09-20T06:15:38.4987800Z  32,681,270 UART: [rt] Received command=0x4d454153, len=108
2023-09-20T06:15:38.4988135Z  32,689,968 >>> mbox cmd response data (12 bytes)
2023-09-20T06:15:38.4988473Z test test_stash_measurement ... FAILED
2023-09-20T06:15:38.4988673Z 
2023-09-20T06:15:38.4988768Z failures:
2023-09-20T06:15:38.4988917Z 
2023-09-20T06:15:38.4989009Z failures:
2023-09-20T06:15:38.4989246Z     test_stash_measurement
2023-09-20T06:15:38.4989424Z 
2023-09-20T06:15:38.4989693Z test result: FAILED. 0 passed; 1 failed; 0 ignored; 0 measured; 18 filtered out; finished in 19932.75s
2023-09-20T06:15:38.4989996Z 
2023-09-20T06:15:38.4990003Z 
2023-09-20T06:15:38.4990524Z --- STDERR:              caliptra-runtime::integration_tests test_stash_measurement ---
2023-09-20T06:15:38.4996100Z thread 'test_stash_measurement' panicked at 'assertion failed: `(left == right)`
2023-09-20T06:15:38.4996619Z   left: `4097`,
2023-09-20T06:15:38.4996995Z  right: `0`', runtime/tests/integration_tests.rs:264:5
```

nquarton, 2023-09-27:
Is anyone looking into this issue yet? If not, I can start

jhand2, 2023-09-27:
@nquarton No I don't think anyone has started on this one. Thanks!

jhand2, 2023-09-27:
Actually, let's split this up into multiple issues so that we can parallelize the work.

I opened https://github.com/chipsalliance/caliptra-sw/issues/882 and assigned to Zach to handle the stuff related to DPE. @nquarton would you want to take the other tests?

* test_keyvault
* test_fw_info
* test_fips_cmd_api

nquarton, 2023-09-29:
> Actually, let's split this up into multiple issues so that we can parallelize the work.
> 
> I opened #882 and assigned to Zach to handle the stuff related to DPE. @nquarton would you want to take the other tests?
> 
> * test_keyvault
> * test_fw_info
> * test_fips_cmd_api

I may not be able to get to these other issues for a week or two due to some other priorities. If no one else can get to them earlier, I could take a look at that point.

nquarton, 2023-09-29:
For this issue, the problem is verilator (and it looks like the FPGA) use a hardcoded PAUSER of 0xFFFF_FFFF. DPE expects a PAUSER of 0x1 from what's provided in the image header. This causes the DPE code to report a failure.

However PAUSER values other than 0xFFFF_FFFF are blocked by the RTL. Only 0xFFFF_FFFF is valid by default (which I'm guessing is an invalid PAUSER? or is this a HW vulnerability?). To allow others, we need to set MBOX_VALID_PAUSER and MBOX_PAUSER_LOCK. I am adding this flow to the hw-model initialization.

We also need to add support to configure the PAUSER used on the bus in all 3 models (it's not configurable in the emulator either).

Ideally, the emulator should be covering this RTL behavior of VALID_PAUSER and PAUSER lock at some point as well.

nquarton, 2023-09-29:
Sorry Jordan, I misunderstood what you were asking at first. I started looking at the stash measurement failure since that was the one that was in the log. I missed the top part that listed all the other failures at first.  If any other DPE failures are for locality/pauser, then the fix I can make should correct those as well.

But still the same answer for the others. Exactly when I can get to them will depend on some other tasks. But I'll put it on my list and I can get to it if no one else gets a chance to.

jhand2, 2023-09-29:
Thanks Nick! No problem, we can look into the other failures and open separate issues.

> To allow others, we need to set MBOX_VALID_PAUSER and MBOX_PAUSER_LOCK. I am adding this flow to the hw-model initialization.
> 
> We also need to add support to configure the PAUSER used on the bus in all 3 models (it's not configurable in the emulator either).

So once you've updated the hw model to set those registers, would you expect this to work for Verilator/FPGA, and then the next step would be to implement use of those registers in the emulator?

cc @zhalvorsen as FYI for why some of the tests are failing.

I think some of the other failures are because the FPGA only supports 8 KeyVaultSlots. I think we use 10 today. We should look into

1. Can DPE use fewer KV slots?
2. If not, can we increase the FPGA to support 16? (FPGA bitstream is really close on space, so we should avoid this if we can)

nquarton, 2023-09-29:
> So once you've updated the hw model to set those registers, would you expect this to work for Verilator/FPGA, and then the next step would be to implement use of those registers in the emulator?

Yep, I pushed a "partial" fix (https://github.com/chipsalliance/caliptra-sw/pull/893) that will change the hardcoded value for PAUSER for verilator and FPGA to 0x1. It also adds a configuration option for the HW-model to init the necessary [MBOX_VALID_PAUSER reg](https://chipsalliance.github.io/caliptra-rtl/main/external-regs/?p=caliptra_top_reg.generic_and_fuse_reg.CPTRA_MBOX_VALID_PAUSER%5B0%5D) with a default of 1 and set LOCK as well. This for sure fixes the stash measurement test on verilator. I haven't had a chance to try the other broken tests or try on FPGA.

The full solution would just add the ability to configure the PAUSER value the 3 models use. Though this could be a separate PR technically since that's really a new feature anyway.

nquarton, 2023-10-03:
Just an update that fix should be complete now. The emulator still doesn't support the VALID_PAUSER check but it also doesn't actually seem to emulate pauser on the bus itself so I think that's an enhancement for a later time. Both VALID_PAUSER and the pauser value we want "soc" to use are configurable now though (and apply to both verilator and FPGA).

zhalvorsen, 2023-10-17:
With #970 added we can now all the caliptra-runtime::integration_tests more easily on the FPGA. `caliptra-runtime::runtime_integration_tests integration_tests::test_update` is now the only test failing.

```
ubuntu@zynqmp:~/caliptra-sw$ CPTRA_UIO_NUM=4 cargo nextest run --no-fail-fast --features=fpga_realtime --release -E 'package(caliptra-runtime)'
    Finished release [optimized] target(s) in 1.57s
    Starting 23 tests across 68 binaries
        PASS [  40.528s] caliptra-runtime::runtime_integration_tests error_handling::test_wdt_timeout
        SLOW [> 60.000s] caliptra-runtime::runtime_integration_tests ecdsa::ecdsa_cmd_run_wycheproof
        SLOW [> 60.000s] caliptra-runtime::runtime_integration_tests hmac::hmac_cmd_run_wycheproof
        SLOW [> 60.000s] caliptra-runtime::runtime_integration_tests integration_tests::test_boot
        PASS [  64.583s] caliptra-runtime::runtime_integration_tests ecdsa::ecdsa_cmd_run_wycheproof
        PASS [  73.007s] caliptra-runtime::runtime_integration_tests hmac::hmac_cmd_run_wycheproof
        PASS [  81.143s] caliptra-runtime::runtime_integration_tests integration_tests::test_boot
        SLOW [> 60.000s] caliptra-runtime::runtime_integration_tests integration_tests::test_certs
        PASS [  71.445s] caliptra-runtime::runtime_integration_tests integration_tests::test_certs
        SLOW [> 60.000s] caliptra-runtime::runtime_integration_tests integration_tests::test_disable_attestation_cmd
        SLOW [> 60.000s] caliptra-runtime::runtime_integration_tests integration_tests::test_ecdsa_verify_cmd
        PASS [  69.842s] caliptra-runtime::runtime_integration_tests integration_tests::test_disable_attestation_cmd
        SLOW [> 60.000s] caliptra-runtime::runtime_integration_tests integration_tests::test_error_cleared
        PASS [  78.127s] caliptra-runtime::runtime_integration_tests integration_tests::test_ecdsa_verify_cmd
        PASS [  87.493s] caliptra-runtime::runtime_integration_tests integration_tests::test_error_cleared
        SLOW [> 60.000s] caliptra-runtime::runtime_integration_tests integration_tests::test_fips_cmd_api
        SLOW [> 60.000s] caliptra-runtime::runtime_integration_tests integration_tests::test_fw_info
        PASS [  87.533s] caliptra-runtime::runtime_integration_tests integration_tests::test_fips_cmd_api
        SLOW [> 60.000s] caliptra-runtime::runtime_integration_tests integration_tests::test_fw_version
        PASS [  90.422s] caliptra-runtime::runtime_integration_tests integration_tests::test_fw_info
        SLOW [> 60.000s] caliptra-runtime::runtime_integration_tests integration_tests::test_idev_id_cert
        PASS [  94.166s] caliptra-runtime::runtime_integration_tests integration_tests::test_fw_version
        SLOW [> 60.000s] caliptra-runtime::runtime_integration_tests integration_tests::test_idev_id_info
        PASS [  94.228s] caliptra-runtime::runtime_integration_tests integration_tests::test_idev_id_cert
        SLOW [> 60.000s] caliptra-runtime::runtime_integration_tests integration_tests::test_invoke_dpe_get_certificate_chain_cmd
        PASS [  93.339s] caliptra-runtime::runtime_integration_tests integration_tests::test_idev_id_info
        SLOW [> 60.000s] caliptra-runtime::runtime_integration_tests integration_tests::test_invoke_dpe_get_profile_cmd
        PASS [  84.675s] caliptra-runtime::runtime_integration_tests integration_tests::test_invoke_dpe_get_certificate_chain_cmd
        SLOW [> 60.000s] caliptra-runtime::runtime_integration_tests integration_tests::test_invoke_dpe_sign_and_certify_key_cmds
        PASS [  94.239s] caliptra-runtime::runtime_integration_tests integration_tests::test_invoke_dpe_get_profile_cmd
        SLOW [> 60.000s] caliptra-runtime::runtime_integration_tests integration_tests::test_pauser_privilege_level_dpe_context_thresholds
        PASS [  93.917s] caliptra-runtime::runtime_integration_tests integration_tests::test_invoke_dpe_sign_and_certify_key_cmds
        SLOW [> 60.000s] caliptra-runtime::runtime_integration_tests integration_tests::test_rt_cert_with_custom_dates
        PASS [  93.420s] caliptra-runtime::runtime_integration_tests integration_tests::test_pauser_privilege_level_dpe_context_thresholds
        PASS [  86.615s] caliptra-runtime::runtime_integration_tests integration_tests::test_rt_cert_with_custom_dates
        SLOW [> 60.000s] caliptra-runtime::runtime_integration_tests integration_tests::test_standard
        SLOW [> 60.000s] caliptra-runtime::runtime_integration_tests integration_tests::test_stash_measurement
        PASS [  86.659s] caliptra-runtime::runtime_integration_tests integration_tests::test_standard
        PASS [  86.096s] caliptra-runtime::runtime_integration_tests integration_tests::test_stash_measurement
        SLOW [> 60.000s] caliptra-runtime::runtime_integration_tests integration_tests::test_unimplemented_cmds
        SLOW [> 60.000s] caliptra-runtime::runtime_integration_tests integration_tests::test_update
        PASS [  43.064s] caliptra-runtime::runtime_integration_tests test_panic_missing::test_panic_missing
        PASS [ 101.508s] caliptra-runtime::runtime_integration_tests integration_tests::test_unimplemented_cmds
        FAIL [ 107.993s] caliptra-runtime::runtime_integration_tests integration_tests::test_update

--- STDOUT:              caliptra-runtime::runtime_integration_tests integration_tests::test_update ---

running 1 test
test integration_tests::test_update has been running for over 60 seconds
[... cut ...]
  = help: consider filing a feature request

thread 'integration_tests::test_update' panicked at 'called `Result::unwrap()` on an `Err` value: MailboxCmdFailed(1048577)', runtime/tests/runtime_integration_tests/integration_tests.rs:78:10
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

------------
     Summary [ 504.465s] 23 tests run: 22 passed (20 slow), 1 failed, 0 skipped
        FAIL [ 107.993s] caliptra-runtime::runtime_integration_tests integration_tests::test_update
error: test run failed
```

jhand2, 2023-10-23:
Closing this issue since there are no failures on the latest FPGA run on Nightly Release: https://github.com/chipsalliance/caliptra-sw/actions/runs/6596754708/job/17923180848

closed 2023-10-23 by jhand2