chipsalliance/caliptra-sw #852: Verilator test failure: caliptra-runtime::ecdsa ecdsa_cmd_run_wycheproof

korran, 2023-09-20:
Looks to be an unexpected NMI error; maybe the watchdog firing?

The runtime should add additional instrumentation to its trap handler to make it easier to diagnose, similar to what the ROM does:

https://github.com/chipsalliance/caliptra-sw/blob/4bacc8fc47402d331d6528e245d673524a27e738/rom/dev/src/main.rs#L248

```
2023-09-20T06:21:48.3612914Z         FAIL [47284.303s] caliptra-runtime::ecdsa ecdsa_cmd_run_wycheproof
2023-09-20T06:21:48.3613428Z 
2023-09-20T06:21:48.3614123Z --- STDOUT:              caliptra-runtime::ecdsa ecdsa_cmd_run_wycheproof ---
2023-09-20T06:21:48.3614583Z 
2023-09-20T06:21:48.3619285Z running 1 test
2023-09-20T06:21:48.3620190Z       6,155 ready_for_fuses is high
2023-09-20T06:21:48.3620920Z       6,333 writing to cptra_bootfsm_go
2023-09-20T06:21:48.3625193Z test ecdsa_cmd_run_wycheproof has been running for over 60 seconds
2023-09-20T06:21:48.3625901Z     195,201 UART: 
2023-09-20T06:21:48.3627019Z     195,256 UART: Running Caliptra ROM ...
2023-09-20T06:21:48.3627755Z     196,662 UART: 
2023-09-20T06:21:48.3628458Z     202,819 UART: [state] CFI Enabled
2023-09-20T06:21:48.3629219Z     205,066 UART: [state] LifecycleState = Unprovisioned
2023-09-20T06:21:48.3639438Z     207,326 UART: [state] DebugLocked = No
2023-09-20T06:21:48.3640557Z     209,534 UART: [state] Watchdog Timer is not started because the device is not locked for debugging
2023-09-20T06:21:48.3642225Z     356,026 UART: ROM Digest: BBB6BA0BFBA2F1E2103A1060EB333FF6613A1F43E352AA823F62EF22650FF80F
2023-09-20T06:21:48.3643161Z     367,266 UART: [kat] ++
2023-09-20T06:21:48.3643755Z     367,778 UART: [kat] sha1
2023-09-20T06:21:48.3644622Z     376,938 UART: [kat] SHA2-256
2023-09-20T06:21:48.3645467Z     379,810 UART: [kat] SHA2-384
2023-09-20T06:21:48.3646078Z     384,075 UART: [kat] SHA2-384-ACC
2023-09-20T06:21:48.3646694Z     386,372 UART: [kat] ECC-384
2023-09-20T06:21:48.3647378Z   3,956,726 UART: [kat] HMAC-384
2023-09-20T06:21:48.3647920Z   3,967,778 UART: [kat] LMS
2023-09-20T06:21:48.3648645Z   5,069,580 UART: [kat] --
2023-09-20T06:21:48.3649427Z   5,072,497 UART: [cold-reset] ++
2023-09-20T06:21:48.3650074Z   5,075,119 UART: [idev] ++
2023-09-20T06:21:48.3650905Z   5,075,698 UART: [idev] CDI.KEYID = 6
2023-09-20T06:21:48.3651663Z   5,076,948 UART: [idev] SUBJECT.KEYID = 7
2023-09-20T06:21:48.3654781Z   5,078,420 UART: [idev] UDS.KEYID = 0
2023-09-20T06:21:48.3656179Z   5,095,715 UART: [idev] Erasing UDS.KEYID = 0
2023-09-20T06:21:48.3657037Z   8,365,879 UART: [idev] Using Sha1 for KeyId Algorithm
2023-09-20T06:21:48.3657549Z   8,387,957 UART: [idev] --
2023-09-20T06:21:48.3657811Z   8,391,516 UART: [ldev] ++
2023-09-20T06:21:48.3658110Z   8,392,109 UART: [ldev] CDI.KEYID = 6
2023-09-20T06:21:48.3658452Z   8,393,351 UART: [ldev] SUBJECT.KEYID = 5
2023-09-20T06:21:48.3658781Z   8,394,823 UART: [ldev] AUTHORITY.KEYID = 7
2023-09-20T06:21:48.3659279Z   8,396,405 UART: [ldev] FE.KEYID = 1
2023-09-20T06:21:48.3659695Z   8,410,196 UART: [ldev] Erasing FE.KEYID = 1
2023-09-20T06:21:48.3660111Z  11,696,727 UART: [ldev] Signing Cert with AUTHORITY.KEYID = 7
2023-09-20T06:21:48.3660707Z  15,292,162 UART: [ldev] PUB.X = 504D38CA45D997901F48BA333A149A2FB2668B973AAC64D3B79ECC09A663F02ED2FABA133F5FA499677AEB7687C99B25
2023-09-20T06:21:48.3661682Z  15,307,860 UART: [ldev] PUB.Y = A5CD31125698B3322C086E4398A591946BF20A3ECEA8EB7D7C23410D49FA877E7ECFBE47BD883BD53D7DA865AC217F1D
2023-09-20T06:21:48.3662439Z  15,323,930 UART: [ldev] SIG.R = 14B5D494DF875AC7CEE9FE3E9A6168A6B64BFB01171F5ECC5216C8BB945FAF957CED1BE6C7CFBB1DDB2B7F4BFD76A492
2023-09-20T06:21:48.3663160Z  15,339,842 UART: [ldev] SIG.S = 821D8CC92F40AFF934771D7652C9BC3C59D47E5635E110A8486B76F7433B02B3C6B0FD8B1B6548B5994C481547111FC5
2023-09-20T06:21:48.3663688Z  15,361,245 UART: [ldev] --
2023-09-20T06:21:48.3663972Z  15,368,037 ready_for_fw is high
2023-09-20T06:21:48.3664331Z  15,368,039 <<< Executing mbox cmd 0x46574c44 (63908 bytes) from SoC
2023-09-20T06:21:48.3664705Z  15,368,098 UART: [fwproc] Waiting for Commands...
2023-09-20T06:21:48.3665089Z  15,404,534 UART: [fwproc] Received command 0x46574c44
2023-09-20T06:21:48.3665481Z  15,407,148 UART: [fwproc] Received Image of size 63908 bytes
2023-09-20T06:21:48.3665881Z  18,126,406 UART: [fwproc] Image verified using Vendor ECC Key Index 0
2023-09-20T06:21:48.3666849Z  18,204,273 UART: [fwproc] Loading FMC at address 0x40000000 len 12964
2023-09-20T06:21:48.3667324Z  18,285,679 UART: [fwproc] Loading Runtime at address 0x40004000 len 45060
2023-09-20T06:21:48.3667684Z  18,560,324 >>> mbox cmd response: success
2023-09-20T06:21:48.3668000Z  18,564,071 UART: [afmc] ++
2023-09-20T06:21:48.3668289Z  18,564,650 UART: [afmc] CDI.KEYID = 6
2023-09-20T06:21:48.3668605Z  18,565,900 UART: [afmc] SUBJECT.KEYID = 7
2023-09-20T06:21:48.3668921Z  18,567,358 UART: [afmc] AUTHORITY.KEYID = 5
2023-09-20T06:21:48.3669301Z  21,880,120 UART: [afmc] Signing Cert with AUTHORITY.KEYID = 5
2023-09-20T06:21:48.3669700Z  24,174,793 UART: [afmc] Erasing AUTHORITY.KEYID = 5
2023-09-20T06:21:48.3670300Z  25,483,075 UART: [afmc] PUB.X = 74A2DA8F43E3CF7B034EB91F9FEBBBEBD3F2BB8B9B15CCF6B2944AE9C548D4ACDD20742598EA46075F5EE3D570A3C884
2023-09-20T06:21:48.3671081Z  25,498,863 UART: [afmc] PUB.Y = F0B6D79EFAF624F13C60ACE8287337FA10B411125D5F76A85844E286F5E554EBD2D1B41B1DFB30A078203A651EFE8C34
2023-09-20T06:21:48.3671785Z  25,514,771 UART: [afmc] SIG.R = 8C94183EECD46495965C1F58565A801C567721535D1E46E8FBD8FE28C4361A17C85DDA0427F22D62A34CAFBA60DE91CF
[... 134 lines cut ...]
2023-09-20T06:21:48.3732125Z  52,276,156 <<< Executing mbox cmd 0x53494756 (196 bytes) from SoC
2023-09-20T06:21:48.3732522Z  52,282,094 UART: [rt] Received command=0x53494756, len=196
2023-09-20T06:21:48.3732866Z  52,285,809 >>> mbox cmd response: failed
2023-09-20T06:21:48.3733246Z  52,285,959 <<< Executing mbox cmd 0x53494756 (196 bytes) from SoC
2023-09-20T06:21:48.3733633Z  52,291,897 UART: [rt] Received command=0x53494756, len=196
2023-09-20T06:21:48.3733967Z  52,295,639 >>> mbox cmd response: failed
2023-09-20T06:21:48.3734341Z  52,295,790 <<< Executing mbox cmd 0x53494756 (196 bytes) from SoC
2023-09-20T06:21:48.3734734Z  52,301,737 UART: [rt] Received command=0x53494756, len=196
2023-09-20T06:21:48.3735141Z  52,671,693 UART: RT NMI mcause=0x00000000 mscause=0x00000000 mepc=0x40005098
2023-09-20T06:21:48.3735515Z  52,673,782 UART: RT Fatal Error: 0x000E000B
2023-09-20T06:21:48.3736192Z  52,675,766 >>> mbox cmd response: failed
2023-09-20T06:21:48.3736536Z test ecdsa_cmd_run_wycheproof ... FAILED
2023-09-20T06:21:48.3736732Z 
2023-09-20T06:21:48.3736828Z failures:
2023-09-20T06:21:48.3736975Z 
2023-09-20T06:21:48.3737067Z failures:
2023-09-20T06:21:48.3737348Z     ecdsa_cmd_run_wycheproof
2023-09-20T06:21:48.3737578Z 
2023-09-20T06:21:48.3737840Z test result: FAILED. 0 passed; 1 failed; 0 ignored; 0 measured; 0 filtered out; finished in 47284.28s
2023-09-20T06:21:48.3738150Z 
2023-09-20T06:21:48.3738156Z 
2023-09-20T06:21:48.3738820Z --- STDERR:              caliptra-runtime::ecdsa ecdsa_cmd_run_wycheproof ---
2023-09-20T06:21:48.3745029Z thread 'ecdsa_cmd_run_wycheproof' panicked at 'called `Result::unwrap()` on an `Err` value: UnableToLockSha512Acc', runtime/tests/ecdsa.rs:80:18
2023-09-20T06:21:48.3745922Z note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

korran, 2023-10-13:
Based on what I'm seeing on the FPGA, this appears to have been fixed by #938

FerralCoder, 2023-11-01:
can we close this?

zhalvorsen, 2023-11-01:
I agree with @korran

closed 2023-11-01 by zhalvorsen