chipsalliance/caliptra-sw #874: ROM Stuck in Entropy_src StartupPhase1

ayanavachakraborty, 2023-09-26:
Hi! We're trying to integrate Caliptra with an SoC. We do see the rom getting stuck looking at the Program Counter it looks like that the rom is stuck at the _StartupPhase1_ state of _entropy_src_main_sm_ module. We're trying to run the Caliptra module in simulation with the SoC and **Internal_TRNG enabled**.

Here is steps that are being following to run the test.

1. Caliptra Rom Image generation:
    a. We followed the examples mentioned here : https://github.com/chipsalliance/caliptra-sw/tree/main/libcaliptra/examples
    b. This generates riscv rom image under caliptra-sw/target/riscv32imc-unknown-none-elf/
    c. We then generated a caliptra-rom.bin from the elf using the standard rust-objcopy command and we loaded that in our simulation environment.
2. HW test env in simluation (Caliptra Data driven by SoC):
    a. Driving the obs_key with some random key data from SoC.
    b. Driving the security_state value to 3'b0 from SoC.
    c. Deassertion of power good and then followed by deassertion of rst_b as mentioned in the Hardware spec document from SoC.
    d. Setting the boot_fsm_go bit from SoC.
    e. Setting all the fuse data from SoC.
     f. Driving the 384bits of itrng data and valid following the etrng_req from SoC (refered from physical_rng.sv).
    g. Diving the Caliptra iTRNG Entropy Configuration 0 reg[31:16] (high threshold) to 0x04C8.
    h. Diving the Caliptra iTRNG Entropy Configuration 0 reg[15:0] (low threshold) to 0x0338.
     i. Driving the Caliptra iTRNG Entropy Configuration 1 reg to 0x14.
     j. Setting the fuse done bit from SoC.
Step 2.d, 2.e and 2.j. are refered from the trace file generated mention on step 1.c.
We've enabled the following switches CALIPTRA_INTERNAL_TRNG, CALIPTRA_SIMULATION, CALIPTRA_INTERNAL_UART.
So, after doing all this step we see the program counter just looping in the entropy_src_main_sm at StartupPhase1.
_After looking in the code we then just realized the trng data to be pushed should be of size 2048bits. Now the state has moved on to StartupPass1 and it stuck there._

**Can someone please guide:**
1. If all the steps are correct that are mentioned above.
2. If any switches are missing to run the simulation.
3. What needs to be done to move ahead and reach to the phase where _ready_for_fw_push_ signal goes high.

_If there is one consolidated document that could guide a user with the steps that he/she needs to following(across all lifecycle stages) that will be really helpful. We faced issues while integrating like preloading the rom the image needs endianess flip, trng data pushed in should be of 2048bits and so._

**We are in a hurry for a release any help here will be highly appreciated.**

-Thanks

JohnTraverAmd, 2023-09-29:
* Can you please add some waveforms?
* Can you confirm that you are compiled for iTRNG?

howardtr, 2023-10-03:
@korran @rkr35 - do we have a rom test for this that @ayanavachakraborty can reference?

@ayanavachakraborty - are you able to provide the items that @JohnTraverAmd  mentioned?  This will help us understand the state / configuration better, thanks!

Note: pls refer to this [doc](https://opentitan.org/book/hw/ip/entropy_src/doc/theory_of_operation.html#theory-of-operation), which describes that the 384 bits are used in the initial boot-time mode to quickly seed the entropy_src.  2048 bits are required when going through the SHA-3 block

rkr35, 2023-10-23:
Apologies for the late response.

You need to supply 4096 bits, which is two windows of 2048 bits.

See [Main State Machine Diagram](https://opentitan.org/book/hw/ip/entropy_src/doc/theory_of_operation.html#main-state-machine-diagram) for more information.

The reason you need to supply two windows is because the startup health check requires two consecutive windows to pass before advancing the state machine to the continuous phase.

FerralCoder, 2023-11-01:
@ayanavachakraborty, can we close this issue now?

korran, 2023-11-10:
Closing as no active on this issue; as far as we can tell the RTL and firmware handle this correctly.

closed 2023-11-10 by korran