chipsalliance/caliptra-sw #883: ROM : Skip the FIPS and KAT tests in DEBUG mode

lemonzinga, 2023-09-28:
In the ROM code, is it possible to skip the FIPS tests, KAT and any other sign/verify operations in DEBUG mode. The motivation for this is to understand the boot process better by brute-forcing the boot stages ROM->FMC->RT( atleast in DEBUG mode).

I'm trying to run a very lean ROM code that does the very basic setup and then moves onto FMC. I want to be able to skip some intensive computational steps without just blanket commenting the code out.

Any help on this would be appreciated.

JohnTraverAmd, 2023-09-29:
https://github.com/chipsalliance/caliptra-sw/pull/710. 

Once this gets merged, then it will be in the next release.

JohnTraverAmd, 2023-10-06:
This looks like it was merged.  Have you confirmed that this has fulfilled your needs?

lemonzinga, 2023-10-06:
The Fake ROM merge fulfills the needs specified here. Thank you! Closing this issue.

closed 2023-10-06 by lemonzinga