chipsalliance/caliptra-sw #884: ROM : ECC-384 waits indefinitely for the VALID bit to be set

lemonzinga, 2023-09-28:
When executing the caliptra-rom code in simulation, specifically the KAT tests and ROM flow function, we see the [ECC-384 code](https://github.com/chipsalliance/caliptra-sw/blob/main/drivers/src/ecc384.rs#L439) waiting indefinitely for the VALID bit to be set. Make note that the simulation sets the caliptra rtl in debug mode. Its not obvious as to what would prompt the ECC engine to come back with a success/failure. Any suggestions on how to debug this would be appreciated.

JohnTraverAmd, 2023-09-29:
Can you specify exactly which ECC operation is hanging?

lemonzinga, 2023-09-29:
@JohnTraverAmd  @mhatrevi 
Certainly,
This is the flow we expect when the IDEVID layer executes.
In the cold reset flow -> **run -> InitDevIdLayer::derive -> generate_csr -> make_csr -> ecdsa384_sign_and_verify -> ecc384.sign -> verify -> verify_r**

This returns from the ECC operation successfully and we were able to verify that it indeed takes about ~1.3 million cycles.

This is the flow we expect when the LDEVID layer executes.
**run -> LocalDevIdLayer::derive -> generate_cert_sig -> ecdsa384_sign_and_verify -> ecc384.sign -> verify -> [verify_r](https://github.com/chipsalliance/caliptra-sw/blob/main/drivers/src/ecc384.rs#L439)**
And it is at this point where its stuck indefinitely for the VALID bit to be set( ~4million cycles and counting ).

Please let us know if there are any pre-requisites that need to satisfied for the LDEVID based ECC op to return successfully. Note : the simulation environment has been setup in DEBUG mode with the security state set to 0x0.

mhatrevi, 2023-10-02:
@lemonzinga There are no pre-requisites for ECC as such. Is it possible to share any debugging info? (console output, waveforms etc.)

JohnTraverAmd, 2023-10-06:
@lemonzinga Are you still having issues here?  If not, can you identify what fixed it?

If you aren't able to respond soon, we may close this.  Feel free to open it back up if there is still an issue.

lemonzinga, 2023-10-06:
We're still having issues on this. I will ask my HW counterpart to comment on this issue and provide more debug info.

mhatrevi, 2023-10-12:
Closing the issue as there haven't been any updates; please open a new issue when additional debug info is available.

closed 2023-10-12 by mhatrevi