chipsalliance/caliptra-sw #903: Ensure ROM testing is complete

FerralCoder, 2023-10-03:
* Review code flows and determine if test is in place
* Develop any identified missing tests
* Use code coverage data to identify any unexpected gaps

andreslagarcavilla, 2023-11-08:
Seems we have a big pile of newly labeled "ROM test" issues. These are blockers for this one?

korran, 2023-11-09:
I created the "ROM Test" issues from the "Test Gaps" list at https://github.com/chipsalliance/caliptra-sw/blob/main/rom/dev/doc/test-coverage/test-coverage.md?plain=1#L126

I don't think anyone should tapeout the ROM until all these gaps are addressed.

FerralCoder, 2023-11-13:
Now that we have the individual tests identified and issues opened to track them, we no longer need this placeholder. I will close this issue