chipsalliance/caliptra-sw #913: Verify that Caliptra is LMS Compliant

JohnTraverAmd, 2023-10-04:
In the IETF RFC 8554 document, there is this line.  Please verify that we are compliant with this or that this doesn't apply.

Section 6, on HSS, page 28 (2nd to last paragraph) states:
HSS allows L=1, in which case the HSS public key and signature formats are essentially the LMS public key and signature formats, prepended by a fixed field. Since HSS with L=1 has very little overhead compared to LMS, all implementations MUST support HSS in order to maximize interoperability.

jhand2, 2023-10-05:
It's a bit hard to tell from reading the code. @ericeilertson can you confirm?

ericeilertson, 2023-10-05:
I don't think we need to worry about HSS.  That RFC is informational.  The NIST ACVP program uses LMS.  CNSA 2.0 specifies all parameters of LMS as being approved. https://nvlpubs.nist.gov/nistpubs/SpecialPublications/NIST.SP.800-208.pdf describes both LMS and HSS.

I think this was the RFC author writing down a thought that never got revisited.  It is only mentioned once on page 28 of the RFC.  I'd also argue with his point about the cost.  There is significant cost to implementing HSS, it doubles the signature size and verification time.  The parsing of HSS vs LMS signatures is terrible since they use the same values in the first parameters, so it is not possible to know what type of signature you are validating.

bluegate010, 2023-10-05:
I wonder if the author meant: "because HSS with L=1 is basically LMS, implementations must support verifying single-level HSS signatures", rather than "implementations must support verifying arbitrary-level HSS signatures".

JohnTraverAmd, 2023-10-09:
It seemed to imply that HSS has a small header on it compared to just LMS.  Is that possible?

ericeilertson, 2023-10-10:
I think your comments about the small header overhead for HSS is likely the correct interpretation.  This HSS header is not in the current NIST ACVP tests nor is it in the reference code the authors provided.  So, I think we can safely ignore it.

FerralCoder, 2023-11-01:
are we ok to close this?

korran, 2023-11-03:
@bluegate010 will followup with some experts.

bluegate010, 2023-11-06:
Have received feedback from multiple sources indicating that neither SP 800-208 nor the latest draft of [ISO 14888-4](https://www.iso.org/standard/80492.html) require rigid compliance with RFC 8554. I think we're good here.

JohnTraverAmd, 2023-11-07:
We also identified the following documentation:

https://csrc.nist.gov/files/pubs/sp/800/208/final/docs/sp800-208-draft-comments-received.pdf

Page 19

Line 438: “This Special Publication approves the use of LMS and HSS…” 
The use of the word “and” implies that NIST approves stand-alone LMS implementations 
that are not themselves HSS with L=1. Section 6 of RFC 8554 states that “Since HSS with 
L=1 has very little overhead compared to LMS, all implementations MUST support HSS in 
order to maximize interoperability”; the somewhat ambiguous language “all 
implementations” is taken to mean “all implementations of LMS”. NIST should make it 
explicit if they wish to allow non-HSS implementations of LMS. However, as LMS is often 
used interchangeably with HSS (which could lead to undue confusion) it is recommended 
that NIST only allow HSS, where single-layer LMS is explicitly HSS. 

**NIST Response: Noted.** The only difference between LMS and HSS with L=1 is the addition 
of some parameter information at the beginning of the encoding for public keys and signatures 
in HSS. As this encoding is outside the scope of SP 800-208, the difference is not relevant to 
this specification. Also, in order to implement the scheme suggested in Section 7 of SP 800-
208, each cryptographic module would implement LMS and an external (non-cryptographic) 
device would use the outputs of these modules to form HSS public keys and signatures. So, 
while relying parties would receive HSS public keys and signatures, the signing cryptographic 
modules would implement LMS, and so the Cryptographic Algorithm Validation Program 
(CAVP) would be validating implementations of LMS key and signature generation.