chipsalliance/caliptra-sw #922: Mitigate glitching of calls within Trng generate

andreslagarcavilla, 2023-10-06:
From Nick Galloway (gallowayn@google.com)

An additional edge case that could be glitched in the early boot sequence before the first random delay happens:

* `rom_entry` calls `caliptra_cfi_lib::cfi_counter::CfiCounter::reset`
  * `reset` calls `caliptra_drivers::trng::Trng::generate`
    * `generate` uses `ureg::read_volatile_slice` and `compiler_builtins::mem::memcpy::memcpy`

Either of which could be attacked with a glitch to trigger a buffer overflow, such as with an instruction skip or bit flip

korran, 2023-10-24:
Reopening until we have a test-case checking that there is no memcpy

korran, 2023-10-24:
Reopening until we have a test-case merged checking that there is no memcpy

PR #989 (2023-10-23): Replace memcpy and read_volatile_slice calls from TRNG with unrolled equivalents
Calling memcpy-like functions before the CFI defenses are initialized is
dangerous, as a glitch could lead to an out-of-bounds write. Fixes #922 

Please review each commit independently.

## Specialize RealMmioMut::read_volatile_array() for risc-v.

These specializations unroll register array reads for 12-word and
16-word reads; they were previously used for RealMmio, but I forgot to
add them for RealMmioMut.

This improves code size, and improves our glitch resistence (since there
is no loop conditional to glitc [... cut]
closed 2023-10-27 by korran