chipsalliance/caliptra-sw #948: ROM - which bin to use?

UriSha1, 2023-10-12:
HI, 
After build Caliptra firmware I did "create cargo".
I got many bin files under  caliptra-sw/target/debug/incremental/
I don't know how to proceed from this point.
1. Which bin file I suppose to use for the ROM?
2. Is there any information of how the bin files are supposed to work?

Thanks

korran, 2023-10-13:
Sorry, we need to make our documentation clearer.

You can get pre-built ROMs from our releases directory:
https://github.com/chipsalliance/caliptra-sw/releases

To build one from scratch, use caliptra-builder:

```
# For a ROM that writes logging characters to generic_output_wires:
cargo run -p caliptra-builder -- --rom-with-log=/tmp/caliptra-rom-with-log.bin

# For a ROM with no logging:
cargo run -p caliptra-builder -- --rom-no-log=/tmp/caliptra-rom.bin
```

UriSha1, 2023-10-15:
Thanks Korran for the reply.
I did the first mentioned command (cargo run -p caliptra-builder -- --rom-with-log=/tmp/caliptra-rom-with-log.bin)
But I got error "error: could not compile `zeroize` (lib)"

I got same error for rom-no-log
Please advice
Thanks

korran, 2023-10-15:
Can you paste the complete output of the command? And also which commit your repo is currently at?

UriSha1, 2023-10-15:
Thanks you for the release link!!
This is very helpful!

Regarding Your questions:
1. Complete error message:
[ushaul@talu63]~/caliptra-sw>cargo run -p caliptra-builder -- --rom-with-log=/tmp/caliptra-rom-with-log.bin
   Compiling proc-macro2 v1.0.66
   Compiling unicode-ident v1.0.8
   Compiling quote v1.0.32
   Compiling syn v1.0.109
   Compiling syn v2.0.28
   Compiling zeroize_derive v1.4.2
   Compiling zerocopy-derive v0.3.2
   Compiling zerocopy v0.6.1
   Compiling zeroize v1.6.0
   Compiling caliptra-lms-types v0.1.0 (/homes/homes_cd1/ushaul/caliptra-sw/lms-types)
   Compiling caliptra-error v0.1.0 (/homes/homes_cd1/ushaul/caliptra-sw/error)
   Compiling caliptra-image-types v0.1.0 (/homes/homes_cd1/ushaul/caliptra-sw/image/types)
   Compiling caliptra-image-gen v0.1.0 (/homes/homes_cd1/ushaul/caliptra-sw/image/gen)
   Compiling openssl-macros v0.1.0
   Compiling openssl v0.10.57
   Compiling caliptra-image-elf v0.1.0 (/homes/homes_cd1/ushaul/caliptra-sw/image/elf)
   Compiling caliptra-image-openssl v0.1.0 (/homes/homes_cd1/ushaul/caliptra-sw/image/openssl)
   Compiling caliptra-image-fake-keys v0.1.0 (/homes/homes_cd1/ushaul/caliptra-sw/image/fake-keys)
   Compiling caliptra-builder v0.1.0 (/homes/homes_cd1/ushaul/caliptra-sw/builder)
    Finished dev [unoptimized + debuginfo] target(s) in 44.98s
     Running `target/debug/image --rom-with-log=/tmp/caliptra-rom-with-log.bin`
/homes/homes_cd1/ushaul/caliptra-sw/builder/../builder/no_meta_rustc_wrapper.sh: line 40: [-1]: bad array subscript
error: could not compile `zeroize` (lib)
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Custom { kind: Other, error: "Process \"/homes/homes_cd1/ushaul/.rustup/toolchains/1.70-x86_64-unknown-linux-gnu/bin/cargo\" CommandArgs { inner: [\"build\", \"--quiet\", \"--locked\", \"--target\", \"riscv32imc-unknown-none-elf\", \"--features\", \"emu,riscv\", \"--no-default-features\", \"--profile\", \"firmware\", \"-p\", \"caliptra-rom\", \"--bin\", \"caliptra-rom\"] } exited with status code Some(101)" }', builder/bin/image_gen.rs:35:82
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
55.866u 8.383s 1:21.04 79.2%	0+0k 656888+726504io 508pf+0w
[ushaul@talu63]~/caliptra-sw>

2. latest commit :commit 0dfb4e5b30feaea975598665ff8074c0d6664426 

3. Cargo Version:
[ushaul@talu63]releases>cargo --version
cargo 1.70.0 (ec8a8a0ca 2023-04-25)

Many Thanks

korran, 2023-10-16:
```
/homes/homes_cd1/ushaul/caliptra-sw/builder/../builder/no_meta_rustc_wrapper.sh: line 40: [-1]: bad array subscript
```

Interesting, apparently this script is using a relatively new bash feature. Can you run `bash --version`?

korran, 2023-10-16:
there's a decent chance #968 will resolve your issue. I recommend upgrading to a linux distribution with a newer copy of bash, though :)

UriSha1, 2023-10-17:
GNU bash, version 4.2.46(2)-release (x86_64-redhat-linux-gnu)
Is that OK?

korran, 2023-10-17:
The script uses a bash feature that was introduced in bash 4.3 (released in 2014). In PR #968, I've replaced that feature with something that is compatible with older shells. Could you try out that PR and see if it solves your problem?

Regardless, you should consider updating your distribution to something with newer tools and libraries; the oldest distribution we test our build with is Ubuntu 20.04 (released in 2020); we make break things for you in the future without our continuous integration noticing. 

Also, for my own reference, may I ask which version of Redhat you're using? If lots of folks are using these older releases, we can consider setting up a CI pipeline to test on one of them.

UriSha1, 2023-10-26:
HI ,
Thanks korran for the reply.

I tried again with another machine which have newer version of bash - GNU bash, version 4.4.20(1)-release (x86_64-redhat-linux-gnu)
I tried the command: cargo run -p caliptra-builder -- --rom-no-log=/tmp/caliptra-rom.bin
I got Warning with error - for example: 
warning: unknown feature specified for `-Ctarget-feature`: `unaligned-scalar-mem`
  |
  = note: it is still passed through to the codegen backend
  = help: consider filing a feature request

error: failed to run custom build command for `openssl-sys v0.9.93`

Caused by:
  process didn't exit successfully: `/homes/homes_cd1/ushaul/caliptra-sw/target/firmware/build/openssl-sys-7fbe011929e60b5f/build-script-main` (exit status: 101)

Any idea what is wrong?
Thanks

UriSha1, 2023-10-29:
Hi, 
Adding that Rust version is :rustc 1.70.0 (90c541806 2023-05-31)
Is that OK?

korran, 2023-11-03:
Cargo will automatically install the correct version, so it doesn't really matter as long as it is reasonable modern (last 6 months?)

UriSha1, 2023-11-05:
HI Korran,
I'm updating that there were some Perl modules missing from the config in our environment.
It looks caliptra-builder is OK, since I'm able to do command for example "cargo test" , but the command 
cargo run -p caliptra-builder -- --rom-no-log=/tmp/caliptra-rom.bin
Still get warning: unknown feature specified for `-Ctarget-feature`: `unaligned-scalar-mem`.

Anyway - I will take the rom from the releases as you previously suggested.
I think this issue can be close , unless you wants to add some comment.

Many Thanks

korran, 2023-11-10:
The warning is expected; resolved in a future version of the rust toolchain that we're not will to upgrade to yet.

closed 2023-11-10 by korran