chipsalliance/caliptra-sw #975: Add support for DMTF otherName in leaf certificate

varuns-nvidia, 2023-10-19:
https://www.dmtf.org/sites/default/files/standards/documents/DSP0274_1.3.0.pdf line 404
This can be provided as a command to runtime with the parameters to ingest. Runtime would update the certificate template (or support a second one) that populates TBS with the information received.

The motivation for this change is to enable Caliptra certificates to be bound to a particular device/model without solely relying on the pCA. A verifier can correlate the device/model in the leaf certificate with the intermediate certificates because they would all have the same UEID.

@jhand2 for thoughts

jhand2, 2023-10-20:
Which cert would this apply to? The DPE leaf cert?

varuns-nvidia, 2023-10-20:
Yep

jhand2, 2023-10-23:
Ok. I think it is reasonable to include this, especially since one of the major DPE use-cases is an an SPDM responder. So would the otherName value be set to the value of the UEID?

By the way, would it still be necessary to do this if DPE leaf certs had the same UEID as the rest of the cert chain? I'm starting to wonder if we're using UEID is too weird of a way and should consider putting the label field somewhere else instead.

PR #1419 (2024-03-29): Add support for custom extensions to the DPE leaf cert
The first custom extension added in this PR is DMTF otherName. 

fixes #975
closed 2024-03-29 by jhand2