chipsalliance/caliptra-sw #982: Fault injection hardening for verifier code

varuns-nvidia, 2023-10-20:
3 asks:
1. Ensure no constant time window between externally-visible signal and critical branch.
2. Add/confirm init-to-fail logic for boolean checks
3. Use multi-bit booleans

Details are captured in https://github.com/chipsalliance/caliptra-sw/blob/main/rom/dev/doc/threat-model.md#mitigation-2

closed 2023-11-08 by mhatrevi

FerralCoder, 2023-11-08:
@mhatrevi Did we cover all the asks here? If not, the remaining items should be split out into new issues.

mhatrevi, 2023-11-08:
> @mhatrevi Did we cover all the asks here? If not, the remaining items should be split out into new issues.

Added issue https://github.com/chipsalliance/caliptra-sw/issues/1057 and https://github.com/chipsalliance/caliptra-sw/issues/1056