chipsalliance/caliptra-sw #1229: FPGA log FIFO overran

jhand2, 2024-01-06:
When running some of the RT tests which perform hitless updates I am hitting https://github.com/chipsalliance/caliptra-sw/blob/809ed5ec8a5f5e963c56190028ed035dca7cf135/hw-model/src/model_fpga_realtime.rs#L264-L266

I'm not sure why this happens locally but does not happen in the CI. I am building the bitstream with itrng enabled if that matters.

In particular, I'm seeing this when running `test_stash_measurement::test_pcr31_extended_upon_stash_measurement`

jhand2, 2024-01-06:
Low priority, just wanted to make a note of it.

jlmahowa-amd, 2024-01-09:
The issue is that for the FPGA realtime model the log fifo is serviced when step() is called. For tests that spend a long time without calling some HwModel interface that calls step() the log fifo can overrun. Since this test builds a second image after the firmware is up it is more likely to hit this.

Tying the log fifo servicing to step() was an easy way to get it to be frequently checked, but isn't sufficient. Moving the log handling to its own thread would remove the assumption that the test is calling step() often.