chipsalliance/caliptra-sw #1234: SHA512 engine: PCR digest breaks PCR extend

ArthurHeymans, 2024-01-08:
https://github.com/chipsalliance/caliptra-sw/pull/1233

Using the PCR digest functionality break extending PCR.
Using zeroize after the digest generation does not fix it either.

jhand2, 2024-01-08:
I was able to repro this on my local FPGA setup. Some additional details:

* Specifically, it looks like taking a PCR quote causes subsequent PCR extensions to extend the incorrect value.
* Calling `gen_pcr_hash` and then reading the PCR values does not modify the PCR state. So it is specifically an interaction with the `extend` functionality.

jhand2, 2024-01-08:
I was also able to reproduce this bug on verilator (took about 4 hours to run the test):

```
 56,831,984 UART: Error: panicked at 'assertion failed: `(left == right)`
 56,834,779 UART:   left: `Array4xN([699224849, 556724558, 4112363357, 3259000261, 2296387275, 125641188, 4210712284, 2439814293, 4071574937, 545824399, 4273271937, 251753514])`,
 56,850,104 UART:  right: `Array4xN([1305145647, 2880066887, 4100731492, 2984532932, 89915406, 1714187402, 751464747, 1143833194, 4038866326, 727270679, 3006259558, 1943085104])`', drivers/test-fw/src/bin/sha384_tests.rs:271:5
```

These are the same values we get on the FPGA, which may point to an RTL bug.

ArthurHeymans, 2024-01-10:
> I was able to repro this on my local FPGA setup. Some additional details:
>
>     * Specifically, it looks like taking a PCR quote causes subsequent PCR extensions to extend the incorrect value.
>
>     * [...]

Actually I found out that also just digests are broken by it.

jhand2, 2025-10-21:
This scared me for a minute cause I thought we forgot to fix it :) Luckily past me to the rescue: https://github.com/chipsalliance/caliptra-sw/pull/1249