chipsalliance/caliptra-sw #1300: RUST not in the correct quality level for RISC-V support

labba, 2024-01-31:
after reviewing the rust platform supported :
https://doc.rust-lang.org/nightly/rustc/platform-support.html

RISC-V is in the Tier 2 level support and according to the documentation:

1. Tier 2 targets can be thought of as "guaranteed to build"
2. Automated tests are not always run so it's **not guaranteed to produce a working build**
3. Tier 2 target-specific code is not closely scrutinized by Rust team(s) when modifications are made. Bugs are possible in all code, **but the level of quality control for these targets is likely to be lower**.

how was Caliptra decided to be written with a CPU that is not a Tier 1 level CPU with this level of compiler quality assurance ?

labba, 2024-03-05:
any updates ? why not share your decision taking ? this is how "transparency" works ? only one side ?

korran, 2024-03-05:
None of the tier-1 supported CPUs are microcontroller-class, so all small embedded Rust projects end up with tier-2 support from the Rust foundation, and must take some responsibility for validating the toolchain themselves.

steven-bellock, 2024-03-05:
See also https://wiki.riseproject.dev/display/HOME/Project+RP004%3A+Support+64-bit+RISC-V+Linux+port+of+Rust+to+Tier-1. Not sure if that got funded though.

labba, 2024-03-10:
> None of the tier-1 supported CPUs are microcontroller-class, so bare-metal Rust projects end up with tier-2 support from the Rust foundation, and must take some responsibility for validating the toolchain themselves.

so "Chips Alliance" is doing all the toolchain validation as well ? what test are they running to get the needed quality ? 

this only shows that Rust is not ready for the bare-metal embedded world since no real commitment from the foundation ...

korran, 2024-03-11:
In general, open-source embedded toolchains tend to be supported mostly by their users. Rust is no exception here.

Bare metal targets are much easier to support than OS-based targets; there is no need to ensure that the standard library supports complex OS APIS such as filesystems and threads. The main concern is rv32imc codegen, which is more of a concern for LLVM than rustc. The founding Caliptra members have many employees devoted to LLVM maintenance, and shouldn't have much trouble finding support if a codegen issue does arise.

As for specific tests, the Caliptra project runs a large set of [caliptra-specific integration tests](https://chipsalliance.github.io/caliptra-sw/index.html) against our binaries, which should notice many codegen issues. We should also consider running any relevant toolchain tests if they were skipped as part of the LLVM or rustc release process.

labba, 2024-03-20:
> In general, open-source embedded toolchains tend to be supported mostly by their users. 

"mostly by their users"  ?? this isn't the right flow, the toolchain suppose to be supported and maintained by the TARGET and not by who use it.. 

In the context of open-source embedded toolchains, the primary responsibility for support and maintenance should lie with the target platform, rather than relying heavily on the users. Specifically, for the RISC-V CPU, the ideal approach is for RISC-V to actively maintain and provide official support for all Rust flavors (such as std and no_std, etc.) within the toolchain. The prevailing industry practice does not typically place the burden of maintenance on end-users.

jhand2, 2024-03-20:
I think in the short term the thing this group has the most control over is doing additional testing and validation ourselves (as @korran mentioned) to ensure Caliptra code and toolchain are as well-validated as possible. I think this is the most constructive way to focus our effort.

In the long term, we definitely would be in support Rust providing tier 1 support for RISC-V. But we also understand that the rust project/maintainers have limited resources (as is true of all toolchains) and it may not be feasible for the Rust project to provide this support at this time.

closed 2024-03-23 by FerralCoder