chipsalliance/caliptra-sw #1313: Crypto functions must Assert that KAT ran in Debug

JohnTraverAmd, 2024-02-05:
Crypto functions must Assert that KAT ran in Debug.

It is required that KATs are run before any crypto functions are used for other purposes.  Thjere should be an assert to ensure that this requirement isn't violated.

It doesn't have to go into production (only check in debug compile) as long as there isn't a different path in production.

swkeypair, 2026-08-18:
Assessed 2026-08-18 by @swkeypair 

Issue: [caliptra-sw issue #1313](https://github.com/chipsalliance/caliptra-sw/issues/1313)

# Supporting evidence 

## 1. ROM runs the common KAT sequence before normal boot flow

ROM invokes `run_fips_tests`, treats an error as fatal, converts the environment to its post-KAT form, and only then enters debug-unlock and normal flow ([`rom/dev/src/main.rs`](https://github.com/chipsalliance/caliptra-sw/blob/main/rom/dev/src/main.rs#L207-L233)). `run_fips_tests` performs the ROM integrity test and invokes the common KAT sequence before reporting completion ([`rom/dev/src/main.rs`](https://github.com/chipsalliance/caliptra-sw/blob/main/rom/dev/src/main.rs#L254-L276)).

This establishes intended boot ordering, but it is not a debug assertion at each cryptographic operation.

## 2. SHA-1 uses a constructor-level KAT guarantee

`Sha1` prevents external direct construction and `Sha1::new()` runs its KAT before returning the driver ([`drivers/src/sha1.rs`](https://github.com/chipsalliance/caliptra-sw/blob/main/drivers/src/sha1.rs#L26-L38)). This is stronger than a debug-only assertion for that driver.

## 3. ROM AES-GCM also uses a constructor-level KAT guarantee

`AesGcm` wraps the underlying AES driver, and its constructor seeds entropy and runs the GCM and CMAC-KDF KATs before returning ([`drivers/src/aes.rs`](https://github.com/chipsalliance/caliptra-sw/blob/main/drivers/src/aes.rs#L1455-L1486)). The only skip-KAT constructor is feature-gated for the emulation-only fake ROM and explicitly marked unsafe ([`drivers/src/aes.rs`](https://github.com/chipsalliance/caliptra-sw/blob/main/drivers/src/aes.rs#L1488-L1505)).

Questions: 
- why is a "CMAC KDF" KAT tied to the AES GCM new? It seems as thought impl AesGcm is also inclusive of CMAC.  
- The AES-256-CMAC KDF (presumably patterned after SP 800-108 KBKDF) is used in the CM_DERIVE_STABLE_KEY process.
An example usage is [cmac_subkey_generation](https://github.com/chipsalliance/Caliptra/tree/main/doc/caliptra_20/fips)  
  - I don't see a [corresponding CAVP certificate](https://github.com/chipsalliance/Caliptra/tree/main/doc/caliptra_20/fips) for that function. 
  - It isn't clear why a different algorithm was needed (CMAC instead of HMAC) for this purpose. 

## 4. LMS records KAT completion, but uses it to avoid duplicate execution

The LMS driver stores a `kat_complete` Boolean ([`drivers/src/lms.rs`](https://github.com/chipsalliance/caliptra-sw/blob/main/drivers/src/lms.rs#L28-L36)). `LmsKat::execute` marks completion, and `execute_once` uses the flag to skip a repeated KAT ([`kat/src/lms_kat.rs`](https://github.com/chipsalliance/caliptra-sw/blob/main/kat/src/lms_kat.rs#L23-L40)).

The cited path does not assert KAT completion at each LMS cryptographic operation.

## 5. General cryptographic operations do not share the requested assertion

For example, ECC-384 `sign` ([`drivers/src/ecc384.rs`](https://github.com/chipsalliance/caliptra-sw/blob/main/drivers/src/ecc384.rs#L618-L651)), `verify` ([`drivers/src/ecc384.rs`](https://github.com/chipsalliance/caliptra-sw/blob/main/drivers/src/ecc384.rs#L669-L689)), and `ecdh` ([`drivers/src/ecc384.rs`](https://github.com/chipsalliance/caliptra-sw/blob/main/drivers/src/ecc384.rs#L764-L790)) are directly callable operations. Their [feature gated fips-test-hooks](https://github.com/chipsalliance/caliptra-sw/blob/main/drivers/src/ecc384.rs#L771) inject test failures; they do not check a KAT-complete state.

The common KAT runner returns an `InitializedDrivers` value containing only the newly constructed SHA-1 driver, while the other drivers remain mutable references in the KAT environment ([`kat/src/lib.rs`](https://github.com/chipsalliance/caliptra-sw/blob/main/kat/src/lib.rs#L47-L60), [`kat/src/lib.rs`](https://github.com/chipsalliance/caliptra-sw/blob/main/kat/src/lib.rs#L120-L129)). There is no common initialized-driver type or debug assertion covering all listed algorithms.

## Conclusions  

Result: The concern appears partially addressed, but additional team review of the issue and this assessment is requested. 

Reason:

- ROM has explicit KAT-before-normal-flow ordering.
- SHA-1 and ROM AES-GCM provide constructor-level KAT guarantees, and LMS tracks completion for one-time KAT execution. Perhaps the style here is a "backstop" to assure that these algorithms are self-tested before use (which might be a good general design principal going forward). 
- The requested general debug assertion is not present across cryptographic driver operations; representative ECC-384 operations do not check KAT completion.

## Recommendations 

The original request is only partially implemented. Keep the issue open unless its scope is narrowed to the drivers with constructor-level guarantees. A complete resolution would use a common post-KAT type/state gate or a debug assertion consistently across approved algorithm entry points without creating a different production execution path.

## Additional comment 
This is a useful feature from an assurance perspective. 
Can a Caliptra block or subsystem be made available for functional testing in the Debug mode, then undergo the process to move to the fielded mode as part of the functional test process?
The Debug mode might also be quite nice to use during code review.

> It doesn't have to go into production (only check in debug compile) as long as there isn't a different path in production.

This could be an explicit step in a functional test plan.

ArthurHeymans, 2026-08-18:
We intended to finish the compile time guarantee that KAT ran for other engines but I guess this was forgotten. I can take a look.