chipsalliance/caliptra-sw #1330: Fix up ROM spec for publication

andreslagarcavilla, 2024-02-09:
@mhatrevi, I was wondering if you can take on this for the ROM spec. We want to complete the spec for project 1p0 milestone

For https://github.com/chipsalliance/caliptra-sw/blob/main/rom/dev/README.md.
* Close the opens in section 2.
* Remove section 5. It is covered in the main spec.
* Remove section 7. It is covered in the hardware spec.
* Make sure section 8 is up-to-date. I think this relates to one of the opens in section 2.
* Resolve TODOs in section 9.6
Thanks, cc @jhand2 @steph-morton

andreslagarcavilla, 2024-02-16:
Hi Vishal, any updates?

mhatrevi, 2024-02-16:
Hi Andres,

I haven't gotten to this yet. I'll work on it today/early next week.

Thanks,
Vishal

andreslagarcavilla, 2024-02-16:
Thanks. Please cc @steph-morton and @jhand2 when you have a PR

andreslagarcavilla, 2024-02-28:
Hi @mhatrevi -- we probably need to get this done this week.

I intend to generate a 1.0 full spec for OCP. Next OCP IC sync is March 7th

mhatrevi, 2024-02-28:
@andreslagarcavilla Started on this.; one PR is out, working on another one for this.

jhand2, 2024-03-04:
For clarity on this issue since section numbers have changed:

* section 2 -> "Spec Opens" (removed)
* Section 5 -> "DICE Layers" (removed)
* Section 7 -> "Vaults" (removed)
* Section 8 -> Firmware Image Bundle (Section 6 as of this writing)
* Section 9.6 -> "Handling commands from Mailbox" (Section 8.5 as of this writing)

The only thing I can't quite tell is if the "Firmware Image Bundle" section is fully updated. @mhatrevi can you confirm if the current state all looks correct?

andreslagarcavilla, 2024-03-04:
Well it's tagged 1.0 now

jhand2, 2024-04-15:
Closing this since we've already compiled all the 1.0 specs

closed 2024-04-15 by jhand2