chipsalliance/caliptra-sw #1410: GitHub releases are vulnerable to insider risk

korran, 2024-03-15:
When releasing a new version, the "Nightly Release' workflow builds the firmware into a zip file and a "GitHub Release" is created for a repo (for example, see the [rt-1.0.0 release](https://github.com/chipsalliance/caliptra-sw/releases/tag/rt-1.0.0)).

Unfortunately, anyone with write access to the caliptra-sw or caliptra-rtl repositories can [modify the released artifacts](https://docs.github.com/en/repositories/releasing-projects-on-github/managing-releases-in-a-repository), without any additional review or audit logging. You can imagine a scenario where an insider modifies the release to add a backdoor, a vendor downloads the release to integrate in their project, and later the insider switches the binary back to the original version to hide the evidence.

Potential solutions:

1. Delete all releases and require vendors to checkout the repo.
2. Publish artifacts into a "caliptra-releases" git repo.
3. Publish artifacts to our [GitHub Pages site](https://chipsalliance.github.io/Caliptra/).

steven-bellock, 2024-03-15:
Presumably 2 and 3 have tighter access controls than the current repositories? Ie, only one or two folks would be able to publish there.

korran, 2024-03-18:
With 2, we would place the artifacts into the git repository itself, which can use branch protection rules to enforce security policy, and the repo itself has a fairly robust audit log (depending how much you trust sha-1). 

With 3, the GHA workflows would be responsible for publishing the binaries, and changing those workflows would require a PR that has to pass the branch protection rules. Although this protects the artifacts from repo writers, the few repo maintainers would still have the ability to change the settings and publish malware to the site.

steven-bellock, 2024-03-31:
This is quite prescient given the xz backdoor.

clundin25, 2025-12-12:
Should be resolved by https://github.com/chipsalliance/caliptra-sw/issues/2706