chipsalliance/caliptra-sw #1424: Add basic compatibility check to FMC

FerralCoder, 2024-03-21:
FMC should add a simple check to verify it is running on a ROM version that it is compatible with. If not, a clear error should be reported to aid in debug.

FerralCoder, 2024-03-21:
Similar issues filed for ROM, FMC, RT:

| Component | Issue Link |
|-|-|
| ROM | https://github.com/chipsalliance/caliptra-sw/issues/1423 |
| FMC | https://github.com/chipsalliance/caliptra-sw/issues/1424 |
| RT | https://github.com/chipsalliance/caliptra-sw/issues/1425 |

leongross, 2025-03-03:
It looks like https://github.com/chipsalliance/caliptra-sw/pull/1503 is an early implementation for this, right?

FerralCoder, 2026-10-24:
Closing as completed. FMC now checks the ROM persistent-data marker and `rom.major_version` at startup (`fmc/src/main.rs`) and halts with the dedicated error `FMC_INVALID_ROM_PERSISTENT_DATA_VERSION` on a mismatch.

Note this check is based on the persistent-data major version rather than the FHT compatibility bits mentioned in the discussion; `fht_major_ver`/`fht_minor_ver` are not validated. If FHT-feature-bit semantics are still wanted, please file a narrower follow-up.

closed 2026-10-24 by FerralCoder