chipsalliance/caliptra-sw #1425: Add basic compatibility check to RT

FerralCoder, 2024-03-21:
RT should add a simple check to verify it is running on an FMC version that it is compatible with. If not, a clear error should be reported to aid in debug.

FerralCoder, 2024-03-21:
Similar issues filed for ROM, FMC, RT:

| Component | Issue Link |
|-|-|
| ROM | https://github.com/chipsalliance/caliptra-sw/issues/1423 |
| FMC | https://github.com/chipsalliance/caliptra-sw/issues/1424 |
| RT | https://github.com/chipsalliance/caliptra-sw/issues/1425 |

benjamindoron, 2024-07-16:
What makes a given stage compatible or incompatible with the one before? Can we know that they're compatible if the two share a major version?

leongross, 2025-02-24:
@jhand2 @FerralCoder have we defined by what we measure compatibility?

jhand2, 2025-02-24:
There are compatibility bits in the FHT that denote support for certain features. So I assume that's what this is referring to

FerralCoder, 2026-10-24:
Closing as completed. Runtime now checks the ROM persistent-data marker, `rom.major_version` and `supports_current_minor_version()`, plus the FW persistent-data marker and `fw.version` (`runtime/src/main.rs`), halting with dedicated `RUNTIME_INVALID_*` error codes on a mismatch.

Note this check is based on the persistent-data versions rather than the FHT compatibility bits mentioned in the discussion; `fht_major_ver`/`fht_minor_ver` are not validated. If FHT-feature-bit semantics are still wanted, please file a narrower follow-up.

closed 2026-10-24 by FerralCoder