chipsalliance/caliptra-sw #1450: Make RT use ROM mailbox driver

sree-revoori1, 2024-04-05:


leongross, 2025-02-03:
@sree-revoori1 could you provide a short description of what you expect here?

jhand2, 2025-02-03:
Currently RT implements its own mailbox driver: https://github.com/chipsalliance/caliptra-sw/blob/main/runtime/src/mailbox.rs. It might be preferable to use the driver that ROM uses instead: https://github.com/chipsalliance/caliptra-sw/blob/main/drivers/src/mailbox.rs

One reason to maybe not do this though: RT FW is very close to the code size limit. The RT mailbox driver is a fair bit simpler and I would assume switching to the common driver would increase code size. So I would only want to switch drivers if the ROM one isn't larger.