chipsalliance/caliptra-sw #1464: Support CI workflow with specified ROM, FMC, and RT versions

jhand2, 2024-04-15:
Revisions of ROM and FMC/RT are spread across different branches/tags. We do not currently have a way to produce a single release which will package binaries at specific revisions (currently our release builds everything from a single commit).

We should be able to build FMC/RT from head, but still bundle ROM for a specific release (e.g. 1.0.1).

This can be a Github workflow which takes as input parameters specific tags/commit hashes at which to build each firmware stage.

benjamindoron, 2024-08-14:
At the moment, we can't build FMC and RT separately: the builder does all of this, then creates the image bundle.
- We could enhance the builder to also support generating a bundle from an FMC and RT ELF

The current plan is to create a new workflow that takes ROM and FW git refs as parameters, and:
- Execute tests: Pass these in to the FPGA and sw-emulator tests. These will now build twice, and copy in the ROM ELF tests as overrides to a destination folder for cargo-nextest to consume.
  - Progress: Done, but many tests still build their own ROM. We need to refactor them to optionally consume a ROM from, for example, a path in an environment variable.
- Build release: Pass these into build_release.sh
  - Progress: Done. But we note that it's fragile, because some paths have moved over the course of releases. For example `hw-latest/` to `hw/latest/`. We might address this in rom-1.0 by symlinking the old path over to the new one. But this will not fix it for any possible git ref.

leongross, 2025-01-24:
Taking over this PR on [leongross/build_versioned_release_rom](https://github.com/chipsalliance/caliptra-sw/tree/leongross/build_versioned_release_rom)

jhand2, 2025-01-24:
Perfect, thanks! Let me know if you have any questions @leongross