chipsalliance/caliptra-sw #1520: RT Cert obtained from the sw-emulator is inconsistent with that on verilator

LuiSzee, 2024-05-20:
Hello, I am doing the fmc::test_hand_off test and found that the RT Cert obtained from the sw-emulator is inconsistent with that on verilator.
Before conducting the test, I made the following modifications in order to ensure that the Caliptra Image remains consistent throughout multiple runs.
Additionally, I have printed the tci value.
```
diff --git a/fmc/src/flow/rt_alias.rs b/fmc/src/flow/rt_alias.rs
index 404925dd..e614a77f 100644
--- a/fmc/src/flow/rt_alias.rs
+++ b/fmc/src/flow/rt_alias.rs
@@ -251,6 +251,8 @@ impl RtAliasLayer {
         let image_manifest_digest: [u8; 48] = okref(&image_manifest_digest)?.into();
         tci[SHA384_HASH_SIZE..2 * SHA384_HASH_SIZE].copy_from_slice(&image_manifest_digest);

+        cprintln!("[tci] {}", HexBytes(&tci));
+
         // Permute CDI from FMC TCI
         Crypto::hmac384_kdf(env, fmc_cdi, b"rt_alias_cdi", Some(&tci), rt_cdi)?;
         report_boot_status(FmcBootStatus::RtAliasDeriveCdiComplete as u32);
diff --git a/fmc/tests/fmc_integration_tests/test_hand_off.rs b/fmc/tests/fmc_integration_tests/test_hand_off.rs
index c6407328..b4c73f29 100644
--- a/fmc/tests/fmc_integration_tests/test_hand_off.rs
+++ b/fmc/tests/fmc_integration_tests/test_hand_off.rs
@@ -1,24 +1,31 @@
 // Licensed under the Apache-2.0 license
 use caliptra_builder::{firmware, ImageOptions};
 use caliptra_hw_model::{BootParams, HwModel, InitParams};
+use std::fs;

 #[test]
 fn test_hand_off() {
     let rom = caliptra_builder::rom_for_fw_integration_tests().unwrap();

-    let image = caliptra_builder::build_and_sign_image(
-        &firmware::FMC_WITH_UART,
-        &firmware::runtime_tests::BOOT,
-        ImageOptions::default(),
-    )
-    .unwrap();
+    //  let image = caliptra_builder::build_and_sign_image(
+    //      &firmware::FMC_WITH_UART,
+    //      &firmware::runtime_tests::BOOT,
+    //      ImageOptions::default(),
+    //  )
+    //  .unwrap();
+
+    //  //println!("{:?}", &image.to_bytes().unwrap());
+    let filename = "/tmp/tmp.bin";
+    //  fs::write(filename, &image.to_bytes().unwrap()).unwrap();
+
+    let read_bytes = fs::read(filename).unwrap();

     let mut hw = caliptra_hw_model::new(BootParams {
         init_params: InitParams {
             rom: &rom,
             ..Default::default()
         },
-        fw_image: Some(&image.to_bytes().unwrap()),
+        fw_image: Some(&read_bytes),
         ..Default::default()
     })
     .unwrap();
diff --git a/kat/src/lms_kat.rs b/kat/src/lms_kat.rs
index 10a0b368..6857a00a 100644
--- a/kat/src/lms_kat.rs
+++ b/kat/src/lms_kat.rs
@@ -332,9 +332,10 @@ impl LmsKat {

         let success =
             lms.verify_lms_signature(sha256_driver, &MESSAGE, &LMS_PUBLIC_KEY, &LMS_SIG)?;
-        if success != LmsResult::Success {
-            Err(CaliptraError::KAT_LMS_DIGEST_MISMATCH)?;
[... 743 lines cut ...]
 21,259,285 UART: [alias rt] Populate DV
 21,261,119 UART: [alias rt] Populate DV Done
 21,266,803 UART: [fht] Handoff : FMC CDI: 6
 21,267,759 UART: [fht] FMC Alias Private Key: 7
 21,269,471 UART: [alias rt] Derive CDI
 21,270,206 UART: [alias rt] Store in in slot 0x4
 21,307,334 UART: [tci] 39B634DCBCA634876A8C53D4DD15A752C9E60E13EB8464F45E0512093F38C3CE04F1E33B3DCBB19902B3FB4F45A729B0E8EC7492CF95F4D9A5167F2C591C995ED187DA3D41BEF5188E88368ABC7DD2F2716A3D5BBFA22CDFFAA30C83A52E288A
 21,326,628 UART: [alias rt] Derive Key Pair
 21,327,529 UART: [alias rt] Store priv key in slot 0x5
 24,568,585 UART: [alias rt] Derive Key Pair - Done
 24,582,264 UART: [alias rt] Signing Cert with AUTHO
 24,583,419 UART:             RITY.KEYID = 7
 26,862,895 UART: [alias rt] Erasing AUTHORITY.KEYID = 7
 26,864,404 UART: [alias rt] PUB.X = 988DF52D19CC213D142F4407C67EEC9257934A45F9312681496970FED206CE035DA841CC523954562C2F06FCE82847AD
 26,871,802 UART: [alias rt] PUB.Y = A2B7C2E885F02E91EB8D51C38FBB117AB29287A1C6296BD047EFEE0724D6FFAF282407C6E320A41DC69C7456B0B7E6FC
 26,879,414 UART: [alias rt] SIG.R = 2C8EB816F05BC5733C4DCF7493CDA41A938CCC29D6186A66A2DF54BAE8B04FCE35DD7CD08870B74DD32EA6B2917E282C
 26,886,852 UART: [alias rt] SIG.S = B7569FCF8B93EF57FC0EDAFAED490EDB502ACFE8DBDF4F7707D8799FAA4F3B0D86951BF7C4EF06DE3DD651F3EDB85F07
 28,191,604 UART: boot::test_boot...    [ok]
* TESTCASE PASSED
test test_hand_off::test_hand_off ... ok

test result: ok. 1 passed; 0 failed; 0 ignored; 0 measured; 4 filtered out; finished in 13693.09s

root@vultr:~/caliptra-sw#
```

jhand2, 2024-05-21:
Thanks @LuiSzee, we're taking a look. I'll update you as soon as I know more.

korran, 2024-05-21:
Incorrect derivation confirmed on FPGA in #1526 :

```
thread 'smoke_test::smoke_test' panicked at 'assertion failed: expected_rt_alias_key.derive_public_key().public_eq(&rt_alias_cert.public_key().unwrap())', test/tests/caliptra_integration_tests/smoke_test.rs:430:5
```

Passes in emulator. I'm pretty sure this is because the FMC is using a 96-byte kdf context which the hardware doesn't like, whereas the ROM uses a 48-byte kdf context (PCR0).

It seems likely that the manifest digest (the second block in the context) is not being mixed into the rt_alias key when running on the FPGA.

Nitsirks, 2024-05-21:
If the block data extends beyond a single block, the key from KV will get cleared from the HMAC and needs to be re-loaded from the KV for the next block iteration. I believe the internal state of HMAC is maintained (we don't zeroize everything) after a KV transaction, so reloading the key should work.

korran, 2024-05-21:
Nitsirks: Where is this procedure documented?

Nitsirks, 2024-05-21:
Apparently it is not. I just checked the hardware specification and keyvault section does not mention the clearing of secrets after use. 

I'll get this updated asap.

This feature came after an ask some time last year to ensure a key from the keyvault could never be used to generate a firmware readable digest, as that could expose device secrets.

PR #1528 (2024-07-01): hmac384 driver: Reload kv keys for each block
[bugfix] hmac384 driver: Reload kv keys for each block.

This fixes #1520 / #1550: HMAC operations larger than asingle block produce the incorrect results when using data or keys from the keyvault.

Additional commits in this PR add driver tests and end-to-end tests that validate this scenario.
commit 500b79dd (2024-07-01): [bugfix] hmac384 driver: Reload kv keys for each block.

This fixes #1520 / #1550: HMAC operations larger than a
single block produce the incorrect results when using data or keys from
the keyvault.
commit e06ee14d (2024-08-15): Caliptra 1.0.3 ROM Updates (#1633)

* Changing SHA384ACC KAT to SHA512 (#1562)

* Changing SHA384ACC KAT to SHA512

- Adding support for SHA512 digest in the SHA acclerator driver
- Enabling support for SHA512 in the SW emulator SHA accelerator

* Renaming sha384acc to sha2_512_384acc

* Adding ECC error checking during waits (#1610)

* [bugfix] hmac384 driver: Reload kv keys for each block.

This fixes #1520 / #1550: HMAC operations larger than a
single block produce the incorrect results when using data or keys from
the keyvault.

Add a multi-block hmac384 test that uses the key-vault.

smok [... cut]
commit a85ffa7d (2024-08-15): [bugfix] hmac384 driver: Reload kv keys for each block.

This fixes #1520 / #1550: HMAC operations larger than a
single block produce the incorrect results when using data or keys from
the keyvault.

Add a multi-block hmac384 test that uses the key-vault.

smoke-test: Verify FMC key derivation with independent impl.
closed 2024-07-01 by jhand2