chipsalliance/caliptra-sw #1559: Caliptra RT FW: Increase number of DPE Context Handles to 32

fdamato, 2024-06-06:
Caliptra RT FW should provide more Context Handles available to the PL0 PAUSER DPE Client. 

Currently, the totla number of Handles are 24: 8 allocated to PL0, 16 allocated to PL1.

Also, given that Caliptra RT FW consumes already 2 localities at boot:
   - Caliptra Locality: for RTFW_Journey PCR
   - PL0 PAUSER Locality: for PAUSER Digest TCI
The effective number of Context Handles available to PL0 Pauser DPE Client is 6.

The suggested solution is to increase the number of Total DPE Context Handles from 24 to 32 and allocate 16 to PL0.

rusty1968, 2024-06-10:
PR available for review.

JohnTraverAmd, 2024-06-10:
https://github.com/chipsalliance/caliptra-sw/pull/1561