chipsalliance/caliptra-sw #1565: Field Entropy potential inconsistency

A-Rovelli, 2024-06-11:
Hello,

We have noticed some differences, or potential inconsistencies, between the fuse map and the corresponding registers, we would need clarification on this. We refer respectively to the tables on the following pages.

Fuse map:
[https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md#fuse-map](https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md#fuse-map)
Fuse and architectural registers
[https://github.com/chipsalliance/caliptra-sw/blob/main/rom/dev/README.md#fuse-and-architectural-registers](https://github.com/chipsalliance/caliptra-sw/blob/main/rom/dev/README.md#fuse-and-architectural-registers)

The first issue is about the size of the field entropy:
map: FIELD ENTROPY (obfuscated) 256 bits
reg: FUSE_FIELD_ENTROPY 384 bits

Thanks in advance.

jhand2, 2024-06-18:
I believe the main spec is correct (256 bits). That also aligns with https://chipsalliance.github.io/caliptra-rtl/main/external-regs/?p=caliptra_top_reg.generic_and_fuse_reg.fuse_field_entropy%5B0%5D. @mhatrevi to confirm.

jhand2, 2024-06-18:
Ah, I see Caleb already confirmed in https://github.com/chipsalliance/Caliptra/issues/192. Thanks, we'll make this update in the ROM spec.