chipsalliance/caliptra-sw #1569: Caliptra 1.0.3 Release

nquarton, 2024-06-13:
Documenting steps taken for the 1.0.2 release here so they can be repeated when 1.0.3 is needed

The latest 1.0.x components are all on the branch caliptra-1.0

- [ ] Create a new branch based on caliptra-1.0
- [ ] Cherry-pick changes labeled with "Caliptra 1.0.x Backport"
- [ ] Add one more commit that:
    - [ ] Updates version numbers in version.rs
    - [ ] Updates RTL submodule version if needed
- [ ] Push and create a PR against the caliptra-1.0 branch
- [ ] Ensure all CI passes and merge the PR (do not squash changes)
- [ ] Manually trigger a nightly release on the caliptra-1.0 branch
NOTE: Publishing the test results to github will fail on any branch other than main. It may make sense to create an html page with the results as an artifact.
- [ ] Re-name release as needed

fdamato, 2024-06-17:
Let`s also update caliptra-dpe to include https://github.com/chipsalliance/caliptra-dpe/commit/3791c34f7002916d776ca906a6fc1d46abbbb163

nquarton, 2024-06-17:
> Let`s also update caliptra-dpe to include [chipsalliance/caliptra-dpe@3791c34](https://github.com/chipsalliance/caliptra-dpe/commit/3791c34f7002916d776ca906a6fc1d46abbbb163)

I've added https://github.com/chipsalliance/caliptra-sw/pull/1575 with the backport label to track this easier.

jhand2, 2024-06-18:
Do you also need https://github.com/chipsalliance/caliptra-dpe/pull/346? That PR looks good to me, just waiting for tests to be fixed.

nquarton, 2024-09-12:
Release is https://github.com/chipsalliance/caliptra-sw/releases/tag/release_v20240826_0

closed 2024-09-12 by nquarton