chipsalliance/caliptra-sw #1664: Behavior for tci_type differs between Stash Measurement calls and Invoke DPE calls for Derive Context

nquarton, 2024-09-04:
We have 3 ways of handling incoming measurements that eventually become a Derive Context call to DPE.

1. Stash measurement called during ROM
2. Stash measurement called during RT
3. RT call to invoke DPE for derive context

In cases 1 and 2 (both stash measurement cases) the tci_type endianness is maintained. The LSB of the type dword is the LSB when it's sent back in the cert.

In case 3, the endianness is seen as flipped in the cert.

In the following test output, each of those scenarios is tried in order. Then, a certify key command is issued and the responding cert is dumped.

ROM Stash (raw cmd payload):
[4, 253, 255, 255, `101, 102, 103, 104`, 1, 2, 3, 4, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 2, 3, 4, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]

RT stash (raw cmd payload):
[220, 252, 255, 255, `101, 102, 103, 104`, 1, 2, 3, 4, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 1, 2, 3, 4, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]

RT invoke DPE - Derive Context (raw cmd payload):
[46, 251, 255, 255, 88, 0, 0, 0, 67, 69, 80, 68, 8, 0, 0, 0, 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 2, 3, 4, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 1, 2, 3, 4, 0, 0, 0, 18, `101, 102, 103, 104`, 0, 0, 0, 0] 

Certify key response cert (tci_type is highlighted, it's just a little hard to see)
[48, 130, 5, 177, 6, 9, 42, 134, 72, 134, 247, 13, 1, 7, 2, 160, 130, 5, 162, 48, 130, 5, 158, 2, 1, 3, 49, 13, 48, 11, 6, 9, 96, 134, 72, 1, 101, 3, 4, 2, 2, 48, 130, 4, 228, 6, 9, 42, 134, 72, 134, 247, 13, 1, 7, 1, 160, 130, 4, 213, 4, 130, 4, 209, 48, 130, 4, 205, 48, 130, 4, 83, 2, 1, 0, 48, 94, 49, 17, 48, 15, 6, 3, 85, 4, 3, 19, 8, 68, 80, 69, 32, 76, 101, 97, 102, 49, 73, 48, 71, 6, 3, 85, 4, 5, 19, 64, 56, 56, 55, 70, 70, 65, 53, 56, 48, 69, 70, 57, 54, 66, 68, 65, 68, 48, 55, 57, 49, 69, 69, 57, 55, 53, 68, 54, 55, 66, 54, 50, 54, 53, 57, 53, 48, 70, 54, 68, 48, 57, 54, 53, 48, 57, 68, 49, 69, 70, 52, 56, 51, 57, 56, 67, 51, 65, 52, 56, 55, 65, 52, 69, 48, 118, 48, 16, 6, 7, 42, 134, 72, 206, 61, 2, 1, 6, 5, 43, 129, 4, 0, 34, 3, 98, 0, 4, 84, 129, 214, 60, 195, 45, 111, 32, 114, 254, 254, 59, 46, 179, 118, 135, 245, 45, 180, 104, 54, 238, 146, 148, 45, 141, 225, 208, 213, 239, 73, 62, 200, 62, 179, 161, 176, 223, 80, 126, 84, 205, 14, 47, 91, 36, 5, 164, 68, 85, 226, 91, 179, 198, 246, 54, 88, 50, 119, 16, 82, 100, 20, 50, 38, 52, 27, 166, 101, 32, 191, 55, 14, 109, 122, 184, 38, 204, 225, 70, 94, 137, 217, 145, 147, 96, 199, 164, 102, 136, 101, 163, 144, 147, 197, 35, 160, 130, 3, 116, 48, 130, 3, 112, 6, 9, 42, 134, 72, 134, 247, 13, 1, 9, 14, 49, 130, 3, 97, 48, 130, 3, 93, 48, 130, 2, 222, 6, 6, 103, 129, 5, 5, 4, 5, 1, 1, 255, 4, 130, 2, 207, 48, 130, 2, 203, 48, 129, 140, 166, 126, 48, 61, 6, 9, 96, 134, 72, 1, 101, 3, 4, 2, 2, 4, 48, 238, 7, 164, 186, 211, 194, 3, 233, 60, 238, 228, 144, 35, 99, 199, 152, 236, 17, 133, 44, 158, 121, 179, 31, 255, 139, 179, 9, 99, 252, 28, 248, 57, 75, 195, 137, 94, 47, 169, 160, 20, 214, 175, 253, 221, 108, 212, 62, 48, 61, 6, 9, 96, 134, 72, 1, 101, 3, 4, 2, 2, 4, 48, 229, 136, 236, 172, 122, 100, 72, 171, 227, 9, 211, 19, 169, 124, 50, 15, 239, 91, 22, 161, 80, 192, 8, 104, 15, 5, 18, 69, 82, 246, 155, 76, 184, 200, 251, 219, 194, 101, 142, 152, 26, 116, 34, 3, 241, 179, 203, 127, 136, 4, 255, 255, 255, 255, 137, 4, 82, 84, 74, 77, 48, 129, 140, 166, 126, 48, 61, 6, 9, 96, 134, 72, 1, 101, 3, 4, 2, 2, 4, 48, 25, 175, 16, 114, 168, 194, 94, 20, 254, 167, 80, 226, 172, 238, 158, 142, 36, 229, 1, 19, 54, 130, 171, 218, 97, 190, 54, 108, 2, 84, 195, 77, 30, 16, 137, 162, 28, 214, 202, 72, 47, 127, 51, 69, 220, 79, 193, 50, 48, 61, 6, 9, 96, 134, 72, 1, 101, 3, 4, 2, 2, 4, 48, 131, 204, 62, 211, 226, 39, 74, 82, 156, 22, 231, 64, 8, 43, 47, 79, 252, 96, 41, 55, 238, 116, 171, 105, 106, 179, 194, 180, 124, 131, 190, 15, 172, 120, 131, 44, 186, 201, 146, 26, 236, 231, 158, 94, 189, 219, 133, 88, 136, 4, 0, 0, 0, 1, 137, 4, 77, 66, 86, 80, 48, 129, 140, 166, 126, 48, 61, 6, 9, 96, 134, 72, 1, 101, 3, 4, 2, 2, 4, 48, 1, 2, 3, 4, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 2, 3, 4, 48, 61, 6, 9, 96, 134, 72, 1, 101, 3, 4, 2, 2, 4, 48, 243, 152, 188, 139, 115, 26, 39, 9, 147, 210, 112, 34, 25, 174, 216, 127, 224, 193, 96, 180, 76, 219, 167, 28, 89, 100, 64, 72, 118, 242, 5, 41, 169, 45, 144, 158, 194, 121, 35, 194, 189, 107, 206, 116, 232, 6, 239, 193, 136, 4, 0, 0, 0, 1, 137, 4, `101, 102, 103, 104`, 48, 129, 140, 166, 126, 48, 61, 6, 9, 96, 134, 72, 1, 101, 3, 4, 2, 2, 4, 48, 1, 2, 3, 4, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 2, 1, 2, 3, 4, 48, 61, 6, 9, 96, 134, 72, 1, 101, 3, 4, 2, 2, 4, 48, 158, 79, 43, 49, 218, 195, 162, 29, 80, 194, 239, 202, 177, 82, 25, 144, 182, 79, 252, 127, 188, 129, 12, 10, 23, 170, 108, 144, 162, 30, 244, 145, 164, 203, 115, 133, 208, 149, 183, 202, 222, 115, 15, 154, 119, 83, 242, 55, 136, 4, 0, 0, 0, 1, 137, 4, `101, 102, 103, 104`, 48, 129, 140, 166, 126, 48, 61, 6, 9, 96, 134, 72, 1, 101, 3, 4, 2, 2, 4, 48, 1, 2, 3, 4, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 1, 2, 3, 4, 48, 61, 6, 9, 96, 134, 72, 1, 101, 3, 4, 2, 2, 4, 48, 85, 18, 56, 119, 246, 230, 111, 223, 154, 129, 241, 131, 165, 244, 54, 192, 96, 249, 61, 48, 171, 230, 85, 230, 219, 161, 29, 14, 139, 129, 20, 161, 19, 14, 212, 79, 160, 202, 38, 205, 208, 59, 20, 240, 107, 75, 107, 199, 136, 4, 0, 0, 0, 1, 137, 4, `104, 103, 102, 101`, 48, 65, 6, 6, 103, 129, 5, 5, 4, 4, 1, 1, 255, 4, 52, 48, 50, 4, 48, 48, 47, 46, 45, 44, 43, 42, 41, 40, 39, 38, 37, 36, 35, 34, 33, 32, 31, 30, 29, 28, 27, 26, 25, 24, 23, 22, 21, 20, 19, 18, 17, 16, 15, 14, 13, 12, 11, 10, 9, 8, 7, 6, 5, 4, 3, 2, 1, 48, 15, 6, 3, 85, 29, 19, 1, 1, 255, 4, 5, 48, 3, 1, 1, 0, 48, 14, 6, 3, 85, 29, 15, 1, 1, 255, 4, 4, 3, 2, 7, 128, 48, 21, 6, 3, 85, 29, 37, 1, 1, 255, 4, 11, 48, 9, 6, 7, 103, 129, 5, 5, 4, 100, 9, 48, 10, 6, 8, 42, 134, 72, 206, 61, 4, 3, 3, 3, 104, 0, 48, 101, 2, 49, 0, 197, 24, 216, 155, 83, 96, 81, 165, 248, 160, 165, 119, 112, 206, 74, 206, 3, 50, 37, 8, 76, 139, 105, 87, 241, 153, 197, 193, 206, 13, 197, 80, 178, 6, 182, 170, 111, 238, 202, 30, 118, 124, 89, 65, 38, 87, 20, 183, 2, 48, 5, 224, 154, 194, 231, 26, 45, 60, 42, 250, 18, 99, 196, 78, 213, 203, 75, 200, 12, 205, 12, 214, 21, 190, 48, 229, 140, 137, 67, 17, 241, 202, 240, 220, 182, 9, 86, 75, 46, 168, 108, 192, 80, 126, 229, 200, 189, 102, 49, 129, 161, 48, 129, 158, 2, 1, 3, 128, 22, 4, 20, 240, 18, 244, 89, 32, 97, 219, 246, 221, 200, 2, 176, 108, 35, 112, 68, 228, 190, 20, 113, 48, 11, 6, 9, 96, 134, 72, 1, 101, 3, 4, 2, 2, 48, 10, 6, 8, 42, 134, 72, 206, 61, 4, 3, 3, 4, 104, 48, 102, 2, 49, 0, 189, 146, 119, 71, 225, 149, 54, 229, 197, 60, 9, 124, 142, 83, 111, 244, 250, 82, 7, 138, 96, 124, 162, 83, 153, 23, 121, 160, 64, 101, 159, 31, 114, 239, 39, 35, 108, 124, 231, 75, 26, 229, 133, 129, 191, 32, 152, 7, 2, 49, 0, 139, 45, 134, 117, 153, 65, 6, 107, 3, 34, 181, 143, 235, 251, 26, 173, 77, 214, 244, 95, 34, 20, 134, 103, 203, 215, 155, 47, 126, 85, 17, 136, 253, 253, 132, 249, 159, 70, 133, 242, 202, 210, 218, 86, 59, 252, 250, 175, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,  [... cut ...] 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]

nquarton, 2024-09-04:
This endianness is flipped in 3 places (when being treated as big endian while being copied)
1. When handling stashed info from ROM. ONLY affects ROM stash measurement. https://github.com/chipsalliance/caliptra-sw/blob/593c9bd2e672421b9ad5f30e8872b57152a5d9d8/runtime/src/drivers.rs#L459
2. When handling a stash measurement call during runtime. ONLY affects runtime stash measurement. https://github.com/chipsalliance/caliptra-sw/blob/593c9bd2e672421b9ad5f30e8872b57152a5d9d8/runtime/src/stash_measurement.rs#L81
3. In DPE when copying to the cert. Affects ALL 3 cases. https://github.com/chipsalliance/caliptra-dpe/blob/f06e88caa2e9d1407160d2fc677293b013fe6a0f/dpe/src/x509.rs#L1249

The result is the endianness is flipped twice for stash measurement calls and thus maintained. In the invoke DPE with derive context call, the endianness is flipped when comparing sending the data vs receiving it.

I would argue all three of these cases where flipping occurs should be removed so the endianness is maintained and no assumptions are being made for the use of this 4 byte field.

jhand2, 2024-09-04:
Thanks @nquarton! I agree with removing all of these cases and just maintaining endianness passed from the caller.

jhand2, 2024-09-04:
Talked with Nick, we should also add a test to ensure that the resulting cert measurement is the same from all 3 ways of making a measurement:

* DeriveContext via INVOKE_DPE command
* STASH_MEASUREMENT called during ROM
* STASH_MEASUREMENT called during RT

closed 2024-10-01 by jhand2