chipsalliance/caliptra-sw #1671: PL0/PL1 Determination for Measurements Stashed in ROM

nquarton, 2024-09-12:
RT currently assumes all measurements passed to ROM are PL0. In a system where the SoC is the only other entity on this bus, the SoC can enforce this. In a shared bus system, this could be an issue.

Changes needed:
- Add feature to ROM to record PAUSER from a stash measurements using 4 bytes reserved in MeasurementLogEntry
- Need to add a capability bit in ROM to indicate if it is using this 4 byte field or not
- RT will check this capability and if enabled, use the 4 byte field to determine PL0/PL1 for the provided measurement

nquarton, 2024-09-17:
What release would we like to target for this fix? 1.2 or backport to 1.1/1.0?

varuns-nvidia, 2024-09-17:
why does a measurement need a PAUSER indication?

nquarton, 2024-09-17:
> why does a measurement need a PAUSER indication?

The PAUSER is used as the locality in the DPE derive context call under the hood. I think the main distinction is that we don't want PL1 users to be able to submit measurements as PL0.

varuns-nvidia, 2024-09-18:
I see. I'd rather the SoC be required to isolate PL0 vs PL1. This is already the semantics defined by STASH_MEASUREMENT:
> STASH_MEASUREMENT

> Makes a measurement into the DPE default context. This command is intendend for callers who update infrequently and cannot tolerate a changing DPE API surface.

>     Call the DPE DeriveContext command with the DefaultContext in the locality of the PL0 PAUSER.

nquarton, 2024-09-18:
Yeah this doesn't affect everyone since in many cases the SoC is the only other user on the bus and can do what you are describing. The concern comes when there are other users on the bus the SoC doesn't have control over.

But I don't think we noticed it explicitly says it uses PL0 in the runtime readme. Either way we have a problem since the runtime and ROM behaviors differ. But we can absolutely continue to discuss what the right way to address this is.

jhand2, 2024-09-18:
Ya, thinking more about this, I think I agree with Varun. It may be better to just require `STASH_MEASUREMENT` calls to come from PL0. The reason is that that allowing other PAUSERs to call it greatly complicates the DPE tree. For example, take the following STASH_MEASUREMENT calls (if we were to allow calls from multiple PAUSERS):

* STASH_MEASUREMENT(m=A) from PAUSER 0
* STASH_MEASUREMENT(m=B) from PAUSER 0
* STASH_MEASUREMENT(m=C) from PAUSER 1
* STASH_MEASUREMENT(m=D) from PAUSER 0

This would result in the following DPE context tree:

[image]

I think this is potentially confusing and probably not what the caller would intend. So probably the best would be to keep the existing behavior:

1. In RT, STASH_MEASUREMENT fails if PAUSER != PL0
2. In ROM, assume that all STASH_MEASUREMENT calls are for PL0.

This does require the chip architecture to only allow PL0 to make STASH_MEASUREMENT calls to ROM (Caliptra ROM cannot enforce this since it doesn't know the PL0 PAUSER until it loads the fw image).

If necessary, we could potentially help the SoC enforce this by adding a ROM API to indicate that ROM STASH_MEASUREMENT calls are done. This would allow SoC PL0 to lock down the STASH_MEASUREMENT interface before letting other cores with APB access to Caliptra out of reset.

fdamato, 2024-09-18:
Hello Jordan,
I agree with you on not allowing Stash Measurement from PL1...don`t see a use case for that either honestly.
I would want also to emphasize Nick`s concern which doesn`t seem to be addressed. The goal of stash measurement API is to be as much as possible stable across ROM / RT FW . 

That being said, what is the concern of Caliptra FW Flagging a ROM "Stashed Measurement" not coming from PL0? Two possible solutions are: 
    - Fatal Error (Maybe too "aggressive")
    - Skipping those from being included in the DPE Context. Error Policy will be then deferred to a verifier which will determine what to do an expected "claim" is missing from the Attestation Report

I would vouch for the second solution, and align Stash Measurement behavior across ROM and Runtime FW

nquarton, 2024-09-18:
Rejecting PL1 calls to stash measurement is perfectly fine with me too. I was just going off of the discussion from last week here.

Maybe this is too messy, but technically the ROM could enforce this without adding an API or changing the handoff to FW. It does get the PL0 PAUSER when it receives the FW. It could just record the PAUSER value from the first stash measurement request then report a fatal error if either:
1. Another stash measurement comes in with a different PAUSER
2. The pl0_pauser in the received (and verified) manifest does not match the recorded PAUSER from observed stash measurement(s)

JohnTraverAmd, 2024-09-18:
How does ROM know what PAUser is PL0 to reject non-PL0 PAUser?  PL0 indication comes from the FW Manifest.

nquarton, 2024-09-18:
> How does ROM know what PAUser is PL0 to reject non-PL0 PAUser? PL0 indication comes from the FW Manifest.

I'm suggesting recording it and deferring the check to when ROM receives the manifest. The only real advantage is we maintain the interface/handoff to RT and contain this change to a single component, ROM. This way RT does not have to determine support multiple ROM behaviors for this.

fdamato, 2024-09-19:
I agree with the advantages of maintaining the interface/handoff to RT and assigning the responsibility of isolating PL0 and PL1 to SOC, as suggested by Varun. This should be a fundamental assumption of the Caliptra 1.x architecture, given that the PL0 value is obtained from the Manifest rather than RTL/Fuses. At this point I`d be inclined to support Jordan's proposal and always assume ROM stash measurements come from PL0

nquarton, 2024-09-19:
If we can always assume/mandate the SoC is the only other user of the bus until Caliptra is out of ROM, then I agree we can skip any extra complexity and assume ROM stash measurements are from PL0.

jhand2, 2024-10-15:
Resolved the gap in RT FW in https://github.com/chipsalliance/caliptra-sw/pull/1704. Any remaining concerns or shall we close this issue?

nquarton, 2024-10-15:
As long as we are good with assuming all ROM stash measurement calls come from PL0 I think this is resolved.

jhand2, 2024-10-15:
Sounds good, I'm going to close this. I think it's reasonable to expect the SoC to boot Caliptra to RT before bringing other cores out of reset. Also the recommended integration has SoC manager sitting in front of Caliptra for all callers anyway.

closed 2024-10-15 by jhand2