chipsalliance/caliptra-sw #1688: Incomplete ITRNG health test documentation

akash-singh-NV, 2024-09-24:
Current spec does not document which ITRNG health tests are enabled by Caliptra ROM in version and how are relevant thresholds calculated and passed from SOC integrator. It is important to document these because any inconsistencies would make Caliptra unusable due to ROM dependencies.

varuns-nvidia, 2024-09-24:
@mhatrevi @jhand2 could you help with documenting the TRNG health tests in the ROM spec?

jhand2, 2024-09-24:
Health test thresholds are set [here](https://github.com/chipsalliance/caliptra-sw/blob/bcc30b1d602bd72f2b857134b3c4021f29394721/drivers/src/csrng.rs#L356) but I don't know how they were chosen. @korran do you know?

The health tests are are done by the IP are briefly described [here](https://opentitan.org/book/hw/ip/entropy_src/index.html).

It looks like we support 2 health checks (the two that are supported by the OpenTitan IP we use):

* Repetition Count Test
* Adaptive Proportion Test

The [documentation](https://opentitan.org/book/hw/ip/entropy_src/index.html) for that IP lists two other checks ("Bucket and Markov tests"), but I don't see anything in ROM that configures them. Is anyone else on the thread able to confirm that these are not enabled?

jhand2, 2024-09-24:
It looks like Kor has also created this PR with some guidance https://github.com/chipsalliance/caliptra-rtl/pull/569. Does this cover what you're looking for?

akash-singh-NV, 2024-09-25:
Kor's PR covers this to an extent. We also need to specify -

1. How ITRNG interprets 4 bit raw noise data - binary symbols vs 4-bit symbols ?
2. RCT implementation in Caliptra 1.x ITRNG + ROM combination is slightly different from expected. Ideally a programmed threshold of N should result in failure when ITRNG sees N identical bits (assuming binary noise source) or N identical 4 bit symbols (assuming non binary 4 bit noise source). However, Caliptra 1.x implementation will flag a failure if ITRNG sees N repetitions on any of the 4 bit lines.

varuns-nvidia, 2024-09-26:
@howardtr for the above questions

akash-singh-NV, 2024-10-23:
Pinging @howardtr

korran, 2024-10-23:
1. How ITRNG interprets 4 bit raw noise data - binary symbols vs 4-bit symbols ?

The ROM is setting [threshold_scope](https://chipsalliance.github.io/caliptra-rtl/main/internal-regs/?p=clp.entropy_src_reg.CONF#THRESHOLD_SCOPE.desc)=TRUE, which IIUC means that all 4 input lines are accumulated into a single score, which should be equivalent to a stream of 1-bit symbols that has been demultiplexed into a 4-bit value. Would need to write tests cases to confirm this understanding, and I don't have time to do that now.

korran, 2024-10-23:
> However, Caliptra 1.x implementation will flag a failure if ITRNG sees N repetitions on any of the 4 bit lines.

Are you sure? The threshold_scope docs indicate otherwise:

https://chipsalliance.github.io/caliptra-rtl/main/internal-regs/?p=clp.entropy_src_reg.CONF#THRESHOLD_SCOPE.desc

akash-singh-NV, 2024-10-24:
> > However, Caliptra 1.x implementation will flag a failure if ITRNG sees N repetitions on any of the 4 bit lines.
> 
> Are you sure? The threshold_scope docs indicate otherwise:
> 
> https://chipsalliance.github.io/caliptra-rtl/main/internal-regs/?p=clp.entropy_src_reg.CONF#THRESHOLD_SCOPE.desc

Yes, looking at the RTL : https://github.com/chipsalliance/caliptra-rtl/blob/main/src/entropy_src/rtl/entropy_src_repcnt_ht.sv.

Also, threshold_scope doesn't apply to RCT. It applies only to APT and Markov tests as per the doc.

akash-singh-NV, 2025-04-08:
Pinging @bharatpillilli @korran for the spec update

moidx, 2025-04-25:
Initial doc updated provided by @korran: https://github.com/chipsalliance/caliptra-rtl/pull/569