chipsalliance/caliptra-sw #1696: DICE: Minor modifications required to TCBInfo Sections of FMC_ALIAS / RT_ALIAS

fdamato, 2024-10-02:
By reviewing the DICE Certificate TCBInfo(s) came across a couple of "minor" issues:

- FMC Alias Certificate Hardcodes Vendor to "Caliptra"   -> Recommendation is to leave this field empty
- FMC Alias Certificate Hardcodes Model to "FMC"   -> Recommendation is to leave this field empty, Field "Type" already contains this info
- FMC Alias Certificate Flags is not properly formatted in the BIT STRING ASN1 type as per [DICE Attestation Architecture Specification](https://trustedcomputinggroup.org/wp-content/uploads/DICE-Attestation-Architecture-Version-1.1-Revision-17_1August2023.pdf)
 
- RT Alias Certificate Hardcodes Vendor to "Caliptra"   -> Recommendation is to leave this field empty
- RT Alias Certificate Hardcodes Model to "RT"   -> Recommendation is to leave this field empty, Field "Type" already contains this info

varuns-nvidia, 2024-10-02:
Do you recommend changing this for 1.x or 2.0?

fdamato, 2024-10-02:
Unless anyone have any objections, I`d target this fix for 1.1.

varuns-nvidia, 2024-10-02:
ROM changes are still open for 1.1?

JohnTraverAmd, 2024-10-02:
A 1.1 ROM went out. Another will have to go out for a few minor fixes. So expect a 1.1.1.

fdamato, 2024-10-18:
One further recommendation is to replace all the DICE Certificate Subject Name from "Caliptra 1.0 .... " to "Caliptra 1.x ....". Currently, both Caliptra 1.0 and Caliptra 1.1 releases show "Caliptra 1.0" in their subject name

nquarton, 2024-10-18:
This should all be addressed in https://github.com/chipsalliance/caliptra-sw/pull/1725

[FMC Alias Cert Before](https://lapo.it/asn1js/#MIIDhzCCAw6gAwIBAgIUZqPnhXyP9dwRAyQfwIZNmxouR1swCgYIKoZIzj0EAwMwaTEcMBoGA1UEAwwTQ2FsaXB0cmEgMS4wIExEZXZJRDFJMEcGA1UEBRNAODA1RjNERTcxM0M5MDhDQTQwRjZFOEUzRjc1MjMwMDVFMzY2MEYxMjg5NTIyQjMzMTI0MkU5MEZGNDYwNTdEOTAiGA8yMDIzMDEwMTAwMDAwMFoYDzk5OTkxMjMxMjM1OTU5WjBsMR8wHQYDVQQDDBZDYWxpcHRyYSAxLjAgRk1DIEFsaWFzMUkwRwYDVQQFE0A2MkEzRTc4NTdDOEZGNURDMTEwMzI0MUZDMDg2NEQ5QjFBMkU0NzVCRUVEQTRDRDI4OUMwOUNGMzRGQ0NDNTM4MHYwEAYHKoZIzj0CAQYFK4EEACIDYgAEAVO76lGMVfYpncsSMJoI1vayhhtGB7FmgXJqzPlobsN7K15kZQo78f4kdrWt1tAeuEa2H0ASBfqZPNIPggthl29NwEVWHR4i_6ieNSFil8syXi_WT06dyc3SrWtaPejBo4IBbjCCAWowEgYDVR0TAQH_BAgwBgEB_wIBAzAOBgNVHQ8BAf8EBAMCAgQwHwYGZ4EFBQQEBBUwEwQRAAAAAAAAAAAAAAAAAAAAAAAwgeIGBmeBBQUEBQSB1zCB1DBygAhDYWxpcHRyYYEGRGV2aWNlgwIBAKY_MD0GCWCGSAFlAwQCAgQwDOHqH_6UqCiKV-GPjbhvJ-PivIMKwh-aPD1ZQ2cguJzDoMksRon1cxJmhCFJ0mQthwUAgAAACYkLREVWSUNFX0lORk-KBQCAAAALMF6ACENhbGlwdHJhgQNGTUODAgEApj8wPQYJYIZIAWUDBAICBDDQtdCQrJ9Y_rtZPoi4uAOzcfPOPElvD9TpXGVCI6rt1OeLmB3nH9qIzHV1U_xOno-JCEZNQ19JTkZPMB0GA1UdDgQWBBRio-eFfI_13BEDJB_Ahk2bGi5HWzAfBgNVHSMEGDAWgBSAXz3nE8kIykD26OP3UjAF42YPEjAKBggqhkjOPQQDAwNnADBkAjBfZNoK7kQ595pdw4k25pYrh1LZu2OKMK_-_MEf5T74BRC2VeDZM7eoRtN9K2rFhR8CMEk2FSVqoD-fn8_U8cHMPKeTdavKWM8HQgtzbXOHRVoTHMiDfnUqUDC13H6MWp5Low)
[FMC Alias Cert After](https://lapo.it/asn1js/#MIIDDTCCApOgAwIBAgIUV2Wkc1AIfJ9CA2YNIP5ww1EgtTYwCgYIKoZIzj0EAwMwbDEfMB0GA1UEAwwWQ2FsaXB0cmEgMS4wIEZNQyBBbGlhczFJMEcGA1UEBRNANjJBM0U3ODU3QzhGRjVEQzExMDMyNDFGQzA4NjREOUIxQTJFNDc1QkVFREE0Q0QyODlDMDlDRjM0RkNDQzUzODAiGA8yMDIzMDEwMTAwMDAwMFoYDzk5OTkxMjMxMjM1OTU5WjBrMR4wHAYDVQQDDBVDYWxpcHRyYSAxLjAgUnQgQWxpYXMxSTBHBgNVBAUTQEQ3NjVBNDczNTAwODdDOUY0MjAzNjYwRDIwRkU3MEMzNTEyMEI1MzZDQUQyMDE0NThCNDZCRThBOTI5NTc1ODUwdjAQBgcqhkjOPQIBBgUrgQQAIgNiAASCM7ciYUCq_qy2HxLhXBamML9Rg2qLj3l5FB-5JKUiMof2d3TBleLw6PSuhNc_5POGumpMq62Z5KX8I4F_usOprtzv2a5s41OqAkfFseocjpFIkK3S98DllQQ8Vy8Q3UGjgfIwge8wEgYDVR0TAQH_BAgwBgEB_wIBAjAOBgNVHQ8BAf8EBAMCAoQwHwYGZ4EFBQQEBBUwEwQRAAAAAAAAAAAAAAAAAAAAAAAwaAYGZ4EFBQQBBF4wXIAIQ2FsaXB0cmGBAlJUgwIBAKY_MD0GCWCGSAFlAwQCAgQwJpivNNRuEz0GE2YHNteFCT7Tup8zA4uk9uT0HbkPx8s-jAofjZj9yBRG0cWynL8ziQdSVF9JTkZPMB0GA1UdDgQWBBTXZaRzUAh8n0IDZg0g_nDDUSC1NjAfBgNVHSMEGDAWgBRio-eFfI_13BEDJB_Ahk2bGi5HWzAKBggqhkjOPQQDAwNoADBlAjAcQ7o3nNVKJA9q-JMoOel_HEgvKKUBRtLwqIVFvxDLHvddKjfN2LmMgNi1MUURemECMQCHHgUvW7DNsVpzesUua40rMBuzvFRZI5SX4PTpnWCyvqOp36C7Gu1W0-K3dU_miQ8)
[RT Alias Cert Before](https://lapo.it/asn1js/#MIIDeTCCAv-gAwIBAgIUXPlHmoGTaSIh5NgRSVjBL0D_vl8wCgYIKoZIzj0EAwMwaTEcMBoGA1UEAwwTQ2FsaXB0cmEgMS54IExEZXZJRDFJMEcGA1UEBRNAODA1RjNERTcxM0M5MDhDQTQwRjZFOEUzRjc1MjMwMDVFMzY2MEYxMjg5NTIyQjMzMTI0MkU5MEZGNDYwNTdEOTAiGA8yMDIzMDEwMTAwMDAwMFoYDzk5OTkxMjMxMjM1OTU5WjBsMR8wHQYDVQQDDBZDYWxpcHRyYSAxLnggRk1DIEFsaWFzMUkwRwYDVQQFE0A1Q0Y5NDc5QTgxOTM2OTIyMjFFNEQ4MTE0OTU4QzEyRjQwRkZCRTVGRUExRjA3NTMwN0E1REIyQjYxQzQ2MUIwMHYwEAYHKoZIzj0CAQYFK4EEACIDYgAEUNBlYzHSMq9-RgMc2aDOWPg5SATp365ImHKdb2Th48HaENny2xlvUrKebWrXTns80aUI-JHvBlKqoN-RXNp7XAxXsoWHE1vKOEV8tYxl4KYCgFS0f9N87pVPMA4WhBqqo4IBXzCCAVswEgYDVR0TAQH_BAgwBgEB_wIBAzAOBgNVHQ8BAf8EBAMCAgQwHwYGZ4EFBQQEBBUwEwQRAAAAAAAAAAAAAAAAAAAAAAAwgdMGBmeBBQUEBQSByDCBADBygAhDYWxpcHRyYYEGRGV2aWNlgwIBxaY_MD0GCWCGSAFlAwQCAgQwDOHqH_6UqCiKV-GPjbhvJ-PivIMKwh-aPD1ZQ2cguJzDoMksRon1cxJmhCFJ0mQthwUAkAAAAYkLREVWSUNFX0lORk-KBQDQAAABME-DAgEApj8wPQYJYIZIAWUDBAICBDCQypNN4o3GGlXkq60XuUNK2EnHLPFDyG1P0gVJYE6XkVod_EQfeyuwuZLTc17vYkyJCEZNQ19JTkZPMB0GA1UdDgQWBBRc-UeagZNpIiHk2BFJWMEvQP--XzAfBgNVHSMEGDAWgBSAXz3nE8kIykD26OP3UjAF42YPEjAKBggqhkjOPQQDAwNoADBlAjBn_SwlxYoTRvZSnPKjMvoW79gegS3iM4FDbPkMZdTr9iavnvMBvTyRt6jfb_MvXNQCMQDLlMMw2pMXvFcE72CivS8d1NRp3lDQdbw_j6GAHrOfE8mROCVA97Nbq6iU5j9tvKc)
[RT Alias Cert After](https://lapo.it/asn1js/#MIIC_jCCAoWgAwIBAgIUD6xpUvFH3k_xMchwCKpdRbKaieYwCgYIKoZIzj0EAwMwbDEfMB0GA1UEAwwWQ2FsaXB0cmEgMS54IEZNQyBBbGlhczFJMEcGA1UEBRNANUNGOTQ3OUE4MTkzNjkyMjIxRTREODExNDk1OEMxMkY0MEZGQkU1RkVBMUYwNzUzMDdBNURCMkI2MUM0NjFCMDAiGA8yMDIzMDEwMTAwMDAwMFoYDzk5OTkxMjMxMjM1OTU5WjBrMR4wHAYDVQQDDBVDYWxpcHRyYSAxLnggUnQgQWxpYXMxSTBHBgNVBAUTQDBCQUM2OTUyRjE0N0RFNEZGMTMxQzg3MDA4QUE1RDQ1QjI5QTg5RTYyMzRGNDlFNUJBOUJBRTU3RDZBNTAxMzcwdjAQBgcqhkjOPQIBBgUrgQQAIgNiAAS1spd57GxxSk0hT-Z5UolvXYLhNYJgCxy9XEwTFhj-QHbfZyoDXlWsxXse6QYuAs40_2JJVwhvrLjUmPb1MX2_wumzOcgRizYSqjubmS60Hx1Cv0IUbfqYPMzfScZZpPqjgeQwgeEwEgYDVR0TAQH_BAgwBgEB_wIBAjAOBgNVHQ8BAf8EBAMCAoQwHwYGZ4EFBQQEBBUwEwQRAAAAAAAAAAAAAAAAAAAAAAAwWgYGZ4EFBQQBBFAwToMCAQCmPzA9BglghkgBZQMEAgIEMCaYrzTUbhM9BhNmBzbXhQk-07qfMwOLpPbk9B25D8fLPowKH42Y_cgURtHFspy_M4kHUlRfSU5GTzAdBgNVHQ4EFgQUC6xpUvFH3k_xMchwCKpdRbKaieYwHwYDVR0jBBgwFoAUXPlHmoGTaSIh5NgRSVjBL0D_vl8wCgYIKoZIzj0EAwMDZwAwZAIwFMI0Gqr90dN0aS-ZEOMnwLn0Mcgg-g1wD75ah_4K8abO8op0bR7ncrgGCYHOqL2VAjB6j27ZwE8BeKBFoXQqB1HrWqcsjGx_ZaJlJL7msP-WhdA1CgYTHyioe6eZ8rJuWa4)

Note this also changes the subject name in IDEV and LDEV certs from "Caliptra 1.0" -> "Caliptra 1.x" as suggested. I did not dump those certs.

nquarton, 2024-10-30:
Updated certs after a few more fixes and getting all the tests to pass.

[FMC Alias Cert Before](https://lapo.it/asn1js/#MIIDhzCCAw6gAwIBAgIUZqPnhXyP9dwRAyQfwIZNmxouR1swCgYIKoZIzj0EAwMwaTEcMBoGA1UEAwwTQ2FsaXB0cmEgMS4wIExEZXZJRDFJMEcGA1UEBRNAODA1RjNERTcxM0M5MDhDQTQwRjZFOEUzRjc1MjMwMDVFMzY2MEYxMjg5NTIyQjMzMTI0MkU5MEZGNDYwNTdEOTAiGA8yMDIzMDEwMTAwMDAwMFoYDzk5OTkxMjMxMjM1OTU5WjBsMR8wHQYDVQQDDBZDYWxpcHRyYSAxLjAgRk1DIEFsaWFzMUkwRwYDVQQFE0A2MkEzRTc4NTdDOEZGNURDMTEwMzI0MUZDMDg2NEQ5QjFBMkU0NzVCRUVEQTRDRDI4OUMwOUNGMzRGQ0NDNTM4MHYwEAYHKoZIzj0CAQYFK4EEACIDYgAEAVO76lGMVfYpncsSMJoI1vayhhtGB7FmgXJqzPlobsN7K15kZQo78f4kdrWt1tAeuEa2H0ASBfqZPNIPggthl29NwEVWHR4i_6ieNSFil8syXi_WT06dyc3SrWtaPejBo4IBbjCCAWowEgYDVR0TAQH_BAgwBgEB_wIBAzAOBgNVHQ8BAf8EBAMCAgQwHwYGZ4EFBQQEBBUwEwQRAAAAAAAAAAAAAAAAAAAAAAAwgeIGBmeBBQUEBQSB1zCB1DBygAhDYWxpcHRyYYEGRGV2aWNlgwIBAKY_MD0GCWCGSAFlAwQCAgQwDOHqH_6UqCiKV-GPjbhvJ-PivIMKwh-aPD1ZQ2cguJzDoMksRon1cxJmhCFJ0mQthwUAgAAACYkLREVWSUNFX0lORk-KBQCAAAALMF6ACENhbGlwdHJhgQNGTUODAgEApj8wPQYJYIZIAWUDBAICBDDQtdCQrJ9Y_rtZPoi4uAOzcfPOPElvD9TpXGVCI6rt1OeLmB3nH9qIzHV1U_xOno-JCEZNQ19JTkZPMB0GA1UdDgQWBBRio-eFfI_13BEDJB_Ahk2bGi5HWzAfBgNVHSMEGDAWgBSAXz3nE8kIykD26OP3UjAF42YPEjAKBggqhkjOPQQDAwNnADBkAjBfZNoK7kQ595pdw4k25pYrh1LZu2OKMK_-_MEf5T74BRC2VeDZM7eoRtN9K2rFhR8CMEk2FSVqoD-fn8_U8cHMPKeTdavKWM8HQgtzbXOHRVoTHMiDfnUqUDC13H6MWp5Low)
[FMC Alias Cert After](https://lapo.it/asn1js/#MIIDeTCCAv-gAwIBAgIUXPlHmoGTaSIh5NgRSVjBL0D_vl8wCgYIKoZIzj0EAwMwaTEcMBoGA1UEAwwTQ2FsaXB0cmEgMS54IExEZXZJRDFJMEcGA1UEBRNAODA1RjNERTcxM0M5MDhDQTQwRjZFOEUzRjc1MjMwMDVFMzY2MEYxMjg5NTIyQjMzMTI0MkU5MEZGNDYwNTdEOTAiGA8yMDIzMDEwMTAwMDAwMFoYDzk5OTkxMjMxMjM1OTU5WjBsMR8wHQYDVQQDDBZDYWxpcHRyYSAxLnggRk1DIEFsaWFzMUkwRwYDVQQFE0A1Q0Y5NDc5QTgxOTM2OTIyMjFFNEQ4MTE0OTU4QzEyRjQwRkZCRTVGRUExRjA3NTMwN0E1REIyQjYxQzQ2MUIwMHYwEAYHKoZIzj0CAQYFK4EEACIDYgAEUNBlYzHSMq9-RgMc2aDOWPg5SATp365ImHKdb2Th48HaENny2xlvUrKebWrXTns80aUI-JHvBlKqoN-RXNp7XAxXsoWHE1vKOEV8tYxl4KYCgFS0f9N87pVPMA4WhBqqo4IBXzCCAVswEgYDVR0TAQH_BAgwBgEB_wIBAzAOBgNVHQ8BAf8EBAMCAgQwHwYGZ4EFBQQEBBUwEwQRAAAAAAAAAAAAAAAAAAAAAAAwgdMGBmeBBQUEBQSByDCBxTBygAhDYWxpcHRyYYEGRGV2aWNlgwIBAKY_MD0GCWCGSAFlAwQCAgQwDOHqH_6UqCiKV-GPjbhvJ-PivIMKwh-aPD1ZQ2cguJzDoMksRon1cxJmhCFJ0mQthwUAkAAAAYkLREVWSUNFX0lORk-KBQDQAAABME-DAgEApj8wPQYJYIZIAWUDBAICBDCQypNN4o3GGlXkq60XuUNK2EnHLPFDyG1P0gVJYE6XkVod_EQfeyuwuZLTc17vYkyJCEZNQ19JTkZPMB0GA1UdDgQWBBRc-UeagZNpIiHk2BFJWMEvQP--XzAfBgNVHSMEGDAWgBSAXz3nE8kIykD26OP3UjAF42YPEjAKBggqhkjOPQQDAwNoADBlAjEAxqvkFK8wVvEuIUzL5pTI13ZA9Dof7yCCD2FGXQEWPOs8xwSsv5Ii-wH0_CjtKmDFAjA_VqqrY1CMd6ZFv2ZBFOsg34TU6hBpRxmPYzXfAz0PJ0YceAotj5oLDiwGMOfcuNM)
[RT Alias Cert Before](https://lapo.it/asn1js/#MIIDDTCCApOgAwIBAgIUTP2vgn5_4jI3CAGp7komzVbxXqQwCgYIKoZIzj0EAwMwbDEfMB0GA1UEAwwWQ2FsaXB0cmEgMS4wIEZNQyBBbGlhczFJMEcGA1UEBRNANjJBM0U3ODU3QzhGRjVEQzExMDMyNDFGQzA4NjREOUIxQTJFNDc1QkVFREE0Q0QyODlDMDlDRjM0RkNDQzUzODAiGA8yMDIzMDEwMTAwMDAwMFoYDzk5OTkxMjMxMjM1OTU5WjBrMR4wHAYDVQQDDBVDYWxpcHRyYSAxLjAgUnQgQWxpYXMxSTBHBgNVBAUTQDQ4RkRBRjgyN0U3RkUyMzIzNzA4MDFBOUVFNEEyNkNENTZGMTVFQTQ3MUFEMzE4OTA0MkQ1MEJFNDUxNjU0NDMwdjAQBgcqhkjOPQIBBgUrgQQAIgNiAAQNEOOcP75JF5sA-f8LpYPkYEh4slcJIBO0QXDjk_B_q6msWbMVm60wFP2dSLzeeo5hJnuvDN_dZT6bpqxh_n3Td3cSLL6HkpIG7sCl0EjmbZKmSo7NAZR8PRKtn1ElMmejgfIwge8wEgYDVR0TAQH_BAgwBgEB_wIBAjAOBgNVHQ8BAf8EBAMCAoQwHwYGZ4EFBQQEBBUwEwQRAAAAAAAAAAAAAAAAAAAAAAAwaAYGZ4EFBQQBBF4wXIAIQ2FsaXB0cmGBAlJUgwIBAKY_MD0GCWCGSAFlAwQCAgQwJpivNNRuEz0GE2YHNteFCT7Tup8zA4uk9uT0HbkPx8s-jAofjZj9yBRG0cWynL8ziQdSVF9JTkZPMB0GA1UdDgQWBBRI_a-Cfn_iMjcIAanuSibNVvFepDAfBgNVHSMEGDAWgBRio-eFfI_13BEDJB_Ahk2bGi5HWzAKBggqhkjOPQQDAwNoADBlAjEA2khbjmIEo850oqz1uFfnQ9U0WogLOPNVNgjNrUxnznxxSPhnm0-K9ysBJZaCRqelAjA4QtBaSWY0vXewj_3VV2_h4GDw80nXyJ8_JK_ivLJwjLcX0ePkEt3dVe-Qh19zvWE)
[RT Alias Cert After](https://lapo.it/asn1js/#MIIC_jCCAoWgAwIBAgIUVPVgIbx3sUFsrc5pVLQX2vZiaGowCgYIKoZIzj0EAwMwbDEfMB0GA1UEAwwWQ2FsaXB0cmEgMS54IEZNQyBBbGlhczFJMEcGA1UEBRNANUNGOTQ3OUE4MTkzNjkyMjIxRTREODExNDk1OEMxMkY0MEZGQkU1RkVBMUYwNzUzMDdBNURCMkI2MUM0NjFCMDAiGA8yMDIzMDEwMTAwMDAwMFoYDzk5OTkxMjMxMjM1OTU5WjBrMR4wHAYDVQQDDBVDYWxpcHRyYSAxLnggUnQgQWxpYXMxSTBHBgNVBAUTQDU0RjU2MDIxQkM3N0IxNDE2Q0FEQ0U2OTU0QjQxN0RBRjY2MjY4NkEzNzhEQUEzMDBDMDFBQzJFQkVFMzIzNzAwdjAQBgcqhkjOPQIBBgUrgQQAIgNiAASCx2ZMIybi3_KFVbcrpo5s6U58NKMxs66HJxjLvctVHNxf9m1IXJZqMWPmEjdp4h4e0h6GZrmp9MqJCcRIiquwDnyUQUXw5l9KkUPkUrJXaOt78n1z6jMLsfyeCbLXxKujgeQwgeEwEgYDVR0TAQH_BAgwBgEB_wIBAjAOBgNVHQ8BAf8EBAMCAoQwHwYGZ4EFBQQEBBUwEwQRAAAAAAAAAAAAAAAAAAAAAAAwWgYGZ4EFBQQBBFAwToMCAQCmPzA9BglghkgBZQMEAgIEMLM3zWpOFkg7ONi7edQgji97ZNO_-Vi9pL4R5YoxOO5150gQRxQcoagQVF4lvksLrYkHUlRfSU5GTzAdBgNVHQ4EFgQUVPVgIbx3sUFsrc5pVLQX2vZiaGowHwYDVR0jBBgwFoAUXPlHmoGTaSIh5NgRSVjBL0D_vl8wCgYIKoZIzj0EAwMDZwAwZAIwMyqRb7g8coZqnaiIDi1N_qAc736-kcdSCV4BBYxB9-xDzDdo1atC_370aPWVpkmWAjAvgfU4KjkQtIW6vdOa6LdeqVD7DidWxD3g6wumEx9Vzf-N0LUJxbTGBRYFzAFY7jo)

closed 2024-11-06 by JohnTraverAmd