chipsalliance/caliptra-sw #1698: Consider deprecating LMS for Caliptra 2.0

bluegate010, 2024-10-02:
If Caliptra 2.0 supports ML-DSA for secure boot, we could strongly push the industry away from stateful key management by simply not supporting LMS as an option for secure boot in Caliptra 2.0.

s32francisco, 2025-02-20:
Does it mean selecting 0x2 in the option for the Firmware Manifest Type field in the Firmware manifest preamble?
Or will Caliptra 2.0 deprecating LMS and only use ML-DSA?
Any decisions or further updates on this discussion?
Since FIPS 204 and FIPS 205 have been released, the implications of LMS support in Caliptra 2.0 seem minimal.

jhand2, 2025-02-26:
Current PoR for 2.0 is to continue to support both options. This is because ML-DSA support is new and we want to support a transition period. Integrators which implemented LMS for 1.x may wish to update to 2.0 without updating their fw signing infrastructure.

We could consider removing LMS support in a future revision.