chipsalliance/caliptra-sw #1735: Add stack overflow detection to SW emulator

clundin25, 2024-10-24:


ArthurHeymans, 2024-10-25:
> https://github.com/chipsalliance/caliptra-sw/pull/1722 uncovered a latent stack overflow. We need to add tests / checks / ci to make sure we are not hitting any stack overflows.

https://github.com/rust-embedded/cortex-m-rt/issues/34 this explains a zero cost abstraction to achieve it

clundin25, 2024-10-25:
That looks like a good approach (cross-linking the tracking bug https://github.com/chipsalliance/caliptra-sw/issues/1665).

Adding stack monitoring in the simulator has a few advantages:
* Doesn't touch any of the prod artifacts.
* Allows for data collection, so we can monitor for code paths that are close to overflowing.

I think it makes sense to implement both.

ArthurHeymans, 2024-10-25:
> That looks like a good approach (cross-linking the tracking bug #1665).
>
> Adding stack monitoring in the simulator has a few advantages:
>
> * Doesn't touch any of the prod artifacts.
> * Allows for data [...]

That makes sense.
Also I think hitless update might be impossible to achieve when changing the stack location. Although I think it's a good idea to do that in 2.0 where the rom is not yet finalized.

closed 2024-12-05 by jhand2