chipsalliance/caliptra-sw #1766: Self Tests Must be Repeated Periodically for FIPS Level 3

nquarton, 2024-11-01:
For FIPS level 3, 7.10.3.8 requires that self tests are repeated periodically.

Note this requirement does allow deferring self tests when critical services are being performed. To reduce latency during these tests, it may make sense to allow the sequence to be interruptible between each test if a mailbox request is received.

nquarton, 2025-01-09:
Open to consider: Is a ROM implementation needed for this? We would not expect ROM to ever be idle long enough for this to run and would prefer to not add anything non-essential in ROM.

nquarton, 2025-03-10:
After speaking with our FIPS consultant again, the suggestion was made to periodically "mark" that the KAT for a particular algorithm needs to re-run instead of actually re-running it every time. Then, the KAT can re=run when that crypto is invoked by other actions. This eliminates wasting time running KATs for crypto that is not being used between these periods.

jhand2, 2025-03-11:
We should consider making the FIPS level configurable (either at compile-time or some runtime mechanism). A device which does not want level 3 support shouldn't have to take the runtime hit of doing this.

closed 2026-11-05 by mhatrevi