chipsalliance/caliptra-sw #1840: Trusted channel meeting FIPS standards needed between Caliptra and SoC

nquarton, 2024-12-06:
A trusted channel is required for communication between the SoC and Caliptra. This channel must protect against tampering and have a physical or logical separation from other ports/interfaces. It must also have an indicator to show when it's being used.

The exact requirements for this are still under discussion. The hope is the isolated AXI bus we have will fulfill these requirements. There is a RFG (request of guidance) pending to address some grey areas and contradictions within the FIPS requirements. https://cmuf-workspace.org/Products/Projects/Tasks.aspx?prjID=19&id=27#

ckavalipati, 2024-12-08:
I am not able see the RFG. I cannot see the project that the above link refers to.

RaunakGu, 2026-09-09:
@nquarton Link seems to be broken, Could you help to update it?