chipsalliance/caliptra-sw #1885: SPDM OIDs not present in DICE certs

bluegate010, 2025-01-09:
[Version 2.6 of the OCP NVMe data center specification](https://www.opencompute.org/documents/datacenter-nvme-ssd-specification-v2-6-2-pdf#page=184) states:

- CERT-5: "The AliasCert chain shall contain an immutable hardware identity certificate and hardware identity OID (refer to the SPDM Specification) shall be used to indicate hardware identity certificate(s)."
- CERT-6: "A Mutable certificate OID (refer to the SPDM Specification) shall be used to indicate mutable certificates."

[SPDM version 1.2](https://www.dmtf.org/sites/default/files/standards/documents/DSP0274_1.2.1.pdf#page=78) defines these OIDs:

- The id-DMTF-hardware-identity OID is defined to help identify the hardware identity certificate in a chain regardless of the certificate chain model used ( DeviceCert or AliasCert ). If the AliasCert model is used, this OID shall not be present in any alias certificates in the chain.
- Mutable certificates may include the id-DMTF-mutable-certificate OID to identify them as mutable. If used, this OID shall be present in all mutable certificates in the chain.

varuns-nvidia, 2025-01-09:
Dupe of https://github.com/chipsalliance/Caliptra/issues/32 ?

RaunakGu, 2026-10-07:
@parvathib, @mhatrevi, Is this still opened items, can you please check these?

</head>

<body lang=EN-US style='tab-interval:.5in;word-wrap:break-word'>

CERT-5 | The   AliasCert chain shall contain an immutable hardware identity certificate and   hardware identity OID (refer to the SPDM Specification) shall be used to   indicate hardware identity certificate(s). |  
-- | -- | --
CERT-6 | A   Mutable certificate OID (refer to the SPDM Specification) shall be used to   indicate mutable certificates. |  

</body>

</html>