chipsalliance/caliptra-sw #1891: Test case needed to retrieve certs with max measurements submitted

nquarton, 2025-01-14:
Issue https://github.com/chipsalliance/caliptra-sw/issues/1832 was about the DPE max cert size being too small for the number of measurements supported. A test case needs to be added that submits all the possible measurements in both contexts and retrieves the cert. Parsing the cert for the measurements is a good idea, but technically DPE returning the cert without error should indicate it was not too large when being generated.

nquarton, 2025-01-15:
This change Anthony is making should enable us to test this easier in the SW emulator: https://github.com/chipsalliance/caliptra-sw/pull/1681

leongross, 2025-04-10:
@nquarton Just to clarify the terminology here: with "max measurements" you mean that we should measure and validate the maximum possible **size** for a x509 certificate that we can create? What are you referring to when you say 

> the number of measurements supported

nquarton, 2025-04-10:
Stash measurement (and derive context) support 16 measurements for each PL0 and PL1. Those all add to the TCB info in the DPE leaf cert and increase its size. The suggestion here was to add a test that stores all 32 possible measurements and then retrieves the DPE leaf cert to ensure there is no error when Caliptra generates and provides it.