chipsalliance/caliptra-sw #1944: Caliptra version related questions

sunguk-bin-fad, 2025-02-05:
1. If Caliptra RTL 1.1 is integrated, must ROM, FMC, RT, and libcaliptra all use version 1.1 (need to use caliptra-1.1 branch)?  
    : Is it possible to use the latest version of ROM, FMC, RT, and libcaliptra? (e.g. ROM1.2, RT1.2,...)  

2. If RTL 1.1 and ROM 1.1 are used, can FMC, RT, and the latest libcaliptra be used?  

3. I am wondering if the reason Caliptra supports Caliptra firmware update is that FMC and RT can be updated regardless of the Caliptra RTL version. If that is not the case, what is the purpose of supporting firmware updates (hitless firmware update)?

jhand2, 2025-02-09:
Latest versions of ROM, FMC, RT, and libcaliptra will all work on 1.1  hardware.

They also work on 1.0 hardware, (we have CI to ensure backward compatibility). Although there is a feature flag on some crates for 1.0 compat.

sunguk-bin-fad, 2025-02-09:
@jhand2 
Thanks for your feedback. I am wondering if ROM, FMC, RT, and libcaliptra always need to maintain the same version. For example, if I use RTL1.1 and ROM1.1, can I use FMC1.2 or the latest version? Likewise, I am curious whether similar dependencies could arise for RT or libcaliptra.

sunguk-bin-fad, 2025-02-19:
> [@jhand2](https://github.com/jhand2) Thanks for your feedback. I am wondering if ROM, FMC, RT, and libcaliptra always need to maintain the same version. For example, if I use RTL1.1 and ROM1.1, can [...]

@jhand2 Can you answer this question?  I am using RTL 1.1 and ROM 1.1, and currently working with the `caliptra-1.1` branch. However, I have noticed that there are bug fixes in the latest branch. Would it be okay to apply the latest branch's RT and libcaliptra all at once?(e.g. main branch's runtime and libcaliptra) or the latest branch's FMC, RT, and libcaliptra? Which approch is available or better?

jhand2, 2025-02-19:
Newer fmc/rt can run on top of older ROMs. Also newer RT can run with older FMC.

libcaliptra is similarly backwards compatible for the most part, however commands added in newer fw versions will fail when run against older rt.

jhand2, 2025-02-19:
I think the simplest option (and the one that's been tested the most) is to use FMC, RT, and libcaliptra all from the same branch. We test this set of FW against all prior ROMs.

FMC has not significantly changed recently, so it would also be supported to use an older FMC with the latest RT/libcaliptra. I would just recommend running all the RT/libcaliptra tests with your specific combination.

sunguk-bin-fad, 2025-02-19:
Thanks for the clarification!