chipsalliance/caliptra-sw #1981: DICE Certificates missing Extended Key Usage OIDs

fdamato, 2025-02-26:
Caliptra DICE Certificates do not include required extensions to indicate the key usage (see TCG DICE Certificate Profile Spec).
   - LDEVID shall contain:
           - tcg-dice-kp-identityLoc (because it is a Local Identity)
           - tcg-dice-kp-eca (because it is an embedded CA)
   - FMC_Alias shall contain:
          - tcg-dice-kp-eca (because it is an embedded CA)
          - tcg-dice-kp-attestLoc (because we use it to sign PCRs)
    - RT_Alias shall contain:
         - tcg-dice-kp-eca (because it is an embedded CA)
- DPE LEAF shall contain:
         - tcg-dice-kp-attestLoc (because we use it to sign PCRs)
-  DPE CA (in case of cdi-export) shall contain
           - tcg-dice-kp-eca (because it is an embedded CA)

timothytrippel, 2025-12-05:
@fdamato could you revisit this to evaluate if this is still applicable?

nquarton, 2025-12-15:
Per Fabrizio, this is still needed for to comply with TCG standards

stellargr, 2026-01-12:
Stella to request an update from Vishal and Fabrizio who should be working on it.