chipsalliance/caliptra-sw #2019: mbox status problem

sunguk-bin-fad, 2025-03-11:
I'm using Caliptra1.1 RTL and FW

In libcaliptra, once caliptra_mailbox_send_complete() is called, mbox_status is always busy. (MBOX_CSR_MBOX_STATUS == 0)  If warm reset is triggered, Caliptra finally disables attestation, because mbox_status is busy.

For example, (check these from caliptra_check_status_get_response())
If mbox_status == CMD_COMPLETE, once clearing mbox_execute mbox_status goes to CMD_BUSY.
If mbox_status == DATA_READY, once clearing mbox_execute mbox_status goes to CMD_BUSY.

I don't think this is normal operation. Is there any history of mbox-related fix? or is there something that I need to check?

sunguk-bin-fad, 2025-03-18:
@jhand2 
Can you check this issue or invite the appropriate person to discuss it? This mailbox issue is reproducible from sw-emulator. After checking-out the main branch, I ran the libcaliptra example application while checking mbox_status both before and after calling caliptra_mbox_write_execute(false).
https://github.com/chipsalliance/caliptra-sw/blob/8b9133a35c3097d91fcebe5ef6d2dfaa9b77d736/libcaliptra/src/caliptra_api.c#L571-L585

before releasing lock : mbox_status (0x102 or 0x101) - CMD_COMPLETE or DATA_READY
after releasing lock : mbox_status (0x0) - which means **CMD_BUSY**

Once warm-reset is triggered, it finally calls DISABLE_ATTESTATION.
https://github.com/chipsalliance/caliptra-sw/blob/8b9133a35c3097d91fcebe5ef6d2dfaa9b77d736/runtime/src/lib.rs#L271-L276

I think either the mailbox status is incorrect(must not be busy), or the condition indicating that Caliptra was running(mbox status = cmd_busy) is wrong. I'd like to hear your opinion on this.

jhand2, 2025-03-18:
mbox_status is only valid while Caliptra is executing a command. Once the lock is released and `mbox_fsm_ps == MBOX_IDLE` then `mbox_status` will reset to 0 (CMD_BUSY).

To determine if the mailbox is busy when initiating the mailbox protocol, the caller should attempt to obtain the lock: https://github.com/chipsalliance/Caliptra/blob/main/doc/caliptra_1x/Caliptra.md#sender-protocol. If `lock` returns 1, then the lock was not granted (different caller holds the lock).

sunguk-bin-fad, 2025-03-18:
> mbox_status is only valid while Caliptra is executing a command. Once the lock is released and `mbox_fsm_ps == MBOX_IDLE` then `mbox_status` will reset to 0 (CMD_BUSY).

From this point, mailbox is idle. So I believe that I shoud be able to trigger warm-reset. However, if I trigger warm-reset, Caliptra runtime FW will disable attesation. (because runtime FW is seeing CMD_BUSY - see the codes below). Are you thinking that we need to fix the condition in the Caliptra runtime FW to avoid seeing CMD_BUSY?
https://github.com/chipsalliance/caliptra-sw/blob/8b9133a35c3097d91fcebe5ef6d2dfaa9b77d736/runtime/src/lib.rs#L271-L276

> To determine if the mailbox is busy when initiating the mailbox protocol, the caller should attempt to obtain the lock: [...]

I believe libcaliptra implements the mailbox protocol, and caliptra_mailbox_send_start() obtains the lock. So there is no issue about this.

sunguk-bin-fad, 2025-03-18:
I’m adding an additional question to clarify the intent of my question.

I thought that DISABLE_ATTESTATION should not be called after a warm reset. In this situation(mailbox FSM = IDLE 
 and mailbox status = CMD_BUSY), is it an intended design for DISABLE_ATTESTATION to be called when a warm reset is triggered?

jhand2, 2025-03-18:
Ah, thanks for clarifying, I understand your question now. I agree that it's incorrect that `DISABLE_ATTESTATION` is called when warm reset is triggered and mailbox is idle.

So I think we should be checking `mbox_fsm_ps` as well to decide if we should disable attestation or not.

sunguk-bin-fad, 2025-03-18:
Thanks for checking. 
- Then, the correct condition is (mbox_fsm_ps != idle) ? 

- Before triggering warm reset, checking (mbox_fsm_ps == idle) is enough? At the end of thread #167 I asked about safe condition to trigger warm reset, and the answer was CPTRA_FLOW_STATUS[mailbox_flow_done] = 1. But mailbox_flow_done was never set to 1.

jhand2, 2025-03-19:
> before triggering warm reset, checking (mbox_fsm_ps == idle) is enough

That is correct

> CPTRA_FLOW_STATUS[mailbox_flow_done] = 1. But mailbox_flow_done was never set to 1.

It looks like we don't set `mailbox_flow_done` anywhere in firmware today. I think we can potentially target that for the next release. But for the 1.2.1 FW release (planned for this week) we're planning to just make the changes in [#2038](https://github.com/chipsalliance/caliptra-sw/pull/2038)

Thanks for finding/reporting this issue!

sunguk-bin-fad, 2025-03-20:
@jhand2
I discussed this issue with our RTL engineer, and @JuneB-Choi found that `mailbox_status` and `mailbox_fsm_ps` are initialized to 0 by a warm reset. In such a case, I believe that checking these two registers in Caliptra FW becomes meaningless. Can you or someone check this ?

The code change below comes from #2038. However, @JuneB-Choi 's analysis is correct, this will not work and DISABLE_ATTESTATION will never be called.
[image]

sunguk-bin-fad, 2025-04-07:
@jhand2 
Do you have any updates or plans regarding this issue? I am curious about how the related discussions are progressing within the community. It seems that PR #2046 has not resolved this issue fully, and I have left comments with some logs that I observed on that PR.

jhand2, 2025-04-07:
@clundin25 can you take a look at https://github.com/chipsalliance/caliptra-sw/pull/2046#discussion_r2030373251?

jhand2, 2025-04-07:
If flow status doesn't persist across warm reset, we may need to use DCCM instead.

clundin25, 2025-04-07:
ACK - I see the mistake in the test that didn't catch this. I agree we will have to use DCCM for this condition.

sunguk-bin-fad, 2025-04-24:
I checked that #2090  resolved this issue. To ensure that the warm reset is triggered while the mailbox is idle, it would be better to acquire the lock before triggering the warm reset. (The lock will be released by the warm reset.)