chipsalliance/caliptra-sw #2070: Failing FPGA tests tracker

clundin25, 2025-04-02:
The following tests are currently failing and marked skipped on the FPGA:

- [ ] caliptra-drivers::drivers_integration_tests test_doe_when_debug_locked
- [ ] caliptra-drivers::drivers_integration_tests test_doe_when_debug_not_locked
- [ ] caliptra-rom::rom_integration_tests test_debug_unlock::test_dbg_unlock_manuf
- [ ] caliptra-rom::rom_integration_tests test_debug_unlock::test_dbg_unlock_manuf_invalid_token
- [ ] caliptra-rom::rom_integration_tests test_debug_unlock::test_dbg_unlock_manuf_passive_mode
- [ ] caliptra-rom::rom_integration_tests test_debug_unlock::test_dbg_unlock_manuf_wrong_cmd
- [ ] caliptra-rom::rom_integration_tests test_debug_unlock::test_dbg_unlock_prod
- [ ] caliptra-rom::rom_integration_tests test_debug_unlock::test_dbg_unlock_prod_invalid_length
- [ ] caliptra-rom::rom_integration_tests test_debug_unlock::test_dbg_unlock_prod_invalid_signature
- [ ] caliptra-rom::rom_integration_tests test_debug_unlock::test_dbg_unlock_prod_invalid_token_challenge
- [ ] caliptra-rom::rom_integration_tests test_debug_unlock::test_dbg_unlock_prod_wrong_cmd
- [ ] caliptra-rom::rom_integration_tests test_debug_unlock::test_dbg_unlock_prod_wrong_public_keys
- [ ] caliptra-rom::rom_integration_tests tests_get_idev_csr::test_validate_csr_mac
- [ ] caliptra-runtime::runtime_integration_tests test_pcr::test_pcr_quote_ecc_and_mldsa
- [ ] caliptra-runtime::runtime_integration_tests test_pcr::test_pcr_quote_ecc_sig_zero
- [ ] caliptra-test::caliptra_integration_tests jtag_test::gdb_test
- [ ] caliptra-test::caliptra_integration_tests smoke_test::retrieve_csr_test
- [ ] caliptra-test::fips_test_suite jtag_locked::jtag_locked

This issue can be resolved once the tests are passing and no longer marked skipped on the FPGA.

clundin25, 2025-04-03:
> jtag_locked - Suspect CI image setup issue. Passes for me.

Here is the log:

```
2025-04-02T17:14:52.4346970Z openocd log returned EOF. Log: sudo: openocd: command not found
```

I agree, the FPGA image is just missing `openocd` and `gdb-multiarch`.

I'll own this task.

jlmahowa-amd, 2025-04-29:
Tested against FPGA built with:
Caliptra RTL:
commit 17a0f47ae3f5722b697579a47044bb6cabae0902 (origin/main, origin/HEAD)
Author: Caleb <11879229+calebofearth@users.noreply.github.com>
Date:   Wed Apr 16 17:44:25 2025 -0700
    [TB] Fix block-size emulation for DMA test across warm resets (#856)
ADB RTL:
commit e754ce5ecc8fe0fbf2ef6fe333ae804f25073303 (HEAD -> main, origin/main, origin/HEAD)
Author: Michael Norris <108370498+Nitsirks@users.noreply.github.com>
Date:   Sun Apr 27 07:57:47 2025 -0700
    [DOC] Updates to clarify memory requirements and streaming message mode (#140)
As of the writing of this caliptra-rtl has 2 new commits, but should be functionally the same.

caliptra-test: test result: ok. 117 passed; 0 failed; 0 ignored; 0 measured; 0 filtered out; finished in 1617.21s
      kat_halt_check_no_output and jtag_locked pass when enabled
      gdb_test fails. Error seems to be when connecting to GDB.

caliptra-drivers: test result: ok. 30 passed; 0 failed; 2 ignored; 0 measured; 0 filtered out; finished in 25.32s
      Ignored tests are mostly config dependent or require the lock-step timing of verilator/sw-emulator. Some ought to be able to be ran on FPGA:
            test_csrng_repetition_count passes
            test_csrng_adaptive_proportion passes
            test_trng_in_itrng_mode passes
            test_trng_in_etrng_mode crashed the FPGA - The FPGA was built for ITRNG so I did expect a failure. This seemed excessive though.

caliptra-rom: test result: FAILED. 138 passed; 13 failed; 4 ignored; 0 measured; 0 filtered out; finished in 401.60
12 tests are related to debug unlock, which is only available in SS mode. This FPGA built without so I believe these failures are expected
    test_debug_unlock::test_dbg_unlock_manuf_invalid_token
    test_debug_unlock::test_dbg_unlock_manuf_req_in_passive_mode
    test_debug_unlock::test_dbg_unlock_manuf_success
    test_debug_unlock::test_dbg_unlock_manuf_wrong_cmd
    test_debug_unlock::test_dbg_unlock_prod_invalid_length
    test_debug_unlock::test_dbg_unlock_prod_invalid_signature
    test_debug_unlock::test_dbg_unlock_prod_invalid_token_challenge
    test_debug_unlock::test_dbg_unlock_prod_success
    test_debug_unlock::test_dbg_unlock_prod_unlock_levels_failure
    test_debug_unlock::test_dbg_unlock_prod_unlock_levels_success
    test_debug_unlock::test_dbg_unlock_prod_wrong_cmd
    test_debug_unlock::test_dbg_unlock_prod_wrong_public_keys
This test initially failed but passed on rerun.
    test_warm_reset::test_warm_reset_during_cold_boot_before_image_validation

caliptra-fmc: test result: ok. 5 passed; 0 failed; 0 ignored; 0 measured; 0 filtered out; finished in 64.53s
caliptra-hw-model: test result: ok. 6 passed; 0 failed; 6 ignored; 0 measured; 0 filtered out; finished in 10.23s