chipsalliance/caliptra-sw #2071: Cumulative TCI doesn't work correctly

sunguk-bin-fad, 2025-04-03:
(Caliptra 1.x)

When adding TCI measurement through DeriveContext, it appears that the cumulative TCI is not being accumulated correctly. As the previous cumulative TCI is initialized to 0, **cumulative TCI = HASH( 0 || current TCI)**. This is because the temporary context does not inherit the cumulative TCI, and the TciNodeData is initialized. (see below) Please check whether this is intended or not.

DeriveContextCmd : 
https://github.com/chipsalliance/caliptra-dpe/blob/54d5d99e567ba64cea5cc3bb2814d2a0697253d0/dpe/src/commands/derive_context.rs#L375-L391

When activating tmp_child_context, tci is initialized to 0 : 
https://github.com/chipsalliance/caliptra-dpe/blob/54d5d99e567ba64cea5cc3bb2814d2a0697253d0/dpe/src/context.rs#L73

And the runtime spec needs to be updated :
From the dpe code, FWID[0] is current TCI
https://github.com/chipsalliance/caliptra-dpe/blob/54d5d99e567ba64cea5cc3bb2814d2a0697253d0/dpe/src/x509.rs#L1239-L1240
Runtime spec indicates that FWID[0] is cumulative TCI
[image]

clundin25, 2025-04-03:
This behavior is correct.

From the "dpe-irot-profile" specification.

> The contexts "Cumulative INPUT_DATA" SHALL be `HASH(DEFAULT_TCI_VALUE || INPUT_DATA)`
        where `DEFAULT_TCI_VALUE` is a Hash-sized buffer of all zeros.

This specification is currently only visible to TCG members.

sunguk-bin-fad, 2025-04-04:
Thanks for letting me know. I'll rephrase the question a bit. 

In DeriveContext, if the recursive flag is set, the tci_cumulative is accumulated with the previous TCI. If the recursive flag is not set, it is accumulated with the DEFAULT_TCI_VALUE (all 0). 

Is it by design that stash measurement does not set the recursive flag when invoking DeriveContext? (It means that stash measurement doesn't support journey TCI)

jhand2, 2025-04-07:
Correct

closed 2025-04-08 by sunguk-bin-fad