chipsalliance/caliptra-sw #2097: Migrate size-history job to new GitHub Cache API

clundin25, 2025-04-14:
GitHub has deprecated the caching API used by the size-history ci-tool.

This API will be removed on April 15th, 2025 so a migration is somewhat urgent.

Without a proper migration, the job will:

1. Delay PRs (Takes > 1 hour to complete).
2. Consume unnecessary compute (Since we are repeating the analysis when we don't have a cache).

Here is an example of a failing job: https://github.com/chipsalliance/caliptra-sw/actions/runs/14337831176/job/40189383098.

Note the error message:

```
Error reading from cache: Unexpected response from server HttpResponse { status: 422, content_type: "application/json; charset=utf-8", data: "{\"$id\":\"1\",\"innerException\":null,\"message\":\"This legacy service is shutting down, effective April 15, 2025. Migrate to the new service ASAP. For more information: [https://gh.io/gha-cache-sunset\](https://gh.io/gha-cache-sunset/)",\"typeName\":\"Microsoft.Azure.DevOps.ArtifactCache.WebApi.ArtifactCacheServiceDecommissionedException, Microsoft.Azure.DevOps.ArtifactCache.WebApi\",\"typeKey\":\"ArtifactCacheServiceDecommissionedException\",\"errorCode\":0,\"eventId\":3000}" }
```

Here is the documentation the failure code leads to: https://github.blog/changelog/2025-03-20-notification-of-upcoming-breaking-changes-in-github-actions/#decommissioned-cache-service-brownouts

clundin25, 2025-04-14:
@MDr164 Can you assign this work item?

MDr164, 2025-04-21:
The new cache API is now protobuf based and fully rewritten so this isn't a small change. Luckily someone already created the new protobuf definitions in rust but that crate requires a newer version of rustc it seems. The output is here: https://github.com/chipsalliance/caliptra-sw/actions/runs/14579608504/job/40893211827?pr=2114

clundin25, 2025-05-16:
I think this is also impacting the sccache logic in the build-test job: https://github.com/mozilla/sccache/issues/2351 and https://github.com/chipsalliance/caliptra-sw/pull/2172.