chipsalliance/caliptra-sw #2121: InitCtxCmd doesn't seem to operate as intended

sunguk-bin-fad, 2025-04-24:
I'd like to check two things related to the DPE context.

1. I'd like to check whether initialize_context(InitCtxCmd) is working as intended. 
- initialize_context can only be called once for default context. But if the SoC calls initialize_context twice with the default flag, both calls succeed. Shouldn't the second call fail?
- Additionally, shouldn't the first call also fail? Caliptra runtime executes initialize_dpe(), and it creates a default context with InitCtxCmd(). Since InitCtxCmd() sets dpe.has_initialized to true, the SoC should fail when invoking initialize_context() with the default flag. However, it does not fail. 

2. Does the context created with derive_context_exported_cdi get erased by a warm reset? and is it intended?
- When deriving a context using derive_context_exported_cdi, if retain_parent flag is not set, DPE validation fails after a warm reset.

clundin25, 2025-04-25:
@sunguk-bin-fad thanks for the questions!

> 1. I'd like to check whether initialize_context(InitCtxCmd) is working as intended

> Additionally, shouldn't the first call also fail? 

I believe it does. Can you share some more details or a repro for this?

The DPE error code is contained in the response header. We should update the libcaliptra suite to check this field.

I patched libcaliptra to check the status field for the response header and we can see that the `InitCtxCmd` fails.

Patch: https://github.com/clundin25/caliptra-sw/commit/51e486e13160752403a04a3969f61ee1239a2527.
Log:

```
  4,491,406 UART: [rt]cmd =0x44504543, len=24
  4,497,885 UART: DPE command failed
DPE Command failed: 0x0
 DPE Error code: 4
Caliptra FW error non-fatal code is 0x0
Caliptra FW error fatal code is 0x0
```

This testcase also implies that the call to initialize_context will fail: https://github.com/chipsalliance/caliptra-sw/blob/main/runtime/tests/runtime_integration_tests/test_invoke_dpe.rs#L169-L191

> Does the context created with derive_context_exported_cdi get erased by a warm reset? and is it intended

The context is stored in DCCM here: https://github.com/chipsalliance/caliptra-sw/blob/main/drivers/src/persistent.rs#L233.

The handle stored in the `Driver` **does** get cleared on a warm reset: https://github.com/chipsalliance/caliptra-sw/blob/main/runtime/src/drivers.rs#L113.

This is **NOT** intended, and we will work on a fix.

> When deriving a context using derive_context_exported_cdi, if retain_parent flag is not set, DPE validation fails after a warm reset.

This is also not intended, and we will also work on a fix for this.

sunguk-bin-fad, 2025-04-28:
@clundin25 
Thanks for checking. I checked that InitCtxCmd with default flag was returning failure. I missed to check the status of DPE response header and just checked the fw_non_fatal / fw_fatal / status.