chipsalliance/caliptra-sw #2137: Skip owner public key check during verify_preamble when drive is in un-provisioned state?

RaunakGu, 2025-05-05:
Owner and vendor public key verification is mandatory. In the Un-provisioned state, the vendor public key check is skipped, but the owner public key check is enforced regardless of the drive state.

1. Shouldn't we also skip the owner public key check when the drive is in the Un-provisioned state?
2. Additionally, owner signature verification should also be skipped when the drive is in the Un-provisioned state.?

[image]

sunguk-bin-fad, 2025-05-12:
From unprovisioned state, vendor/owner PK digest verificaitons are skipped and signature verifications are alwayes enforced. For the owner PK digest verification, it doesn't see the lifecycle state, but it verifies the digest only when the owner PK digest in the fuse.

RaunakGu, 2025-05-15:
Thanks for the reply,
@sunguk-bin-fad 

Can you please confirm verify behaviour if it will work in un-provisioned state?
It seems this is being rejected for pqc_key_type.

[image]

sunguk-bin-fad, 2025-05-15:
This is the first time I've seen the Caliptra2.x code, but since '0' cannot be a valid key type, it is likely to cause issues in the unprovisioned state. (The sw-emulator may set the fuse to the default key type(1) and avoids issues, but this could become a problem on the SoC.)

cc @mhatrevi

mhatrevi, 2025-05-22:
@sunguk-bin-fad  We'll discuss this internally and get back to you.