chipsalliance/caliptra-sw #2145: Request for ROM 1.2 release notes and bug fix summary

Neal-liu, 2025-05-09:
Hi Caliptra team,

I would like to know if there is a release note or a summary document available for ROM version 1.2.
Specifically:
1. Is there an official release note or a list of changes for ROM 1.2?
2. Were there any critical or severe bugs in ROM 1.1 that have been fixed in ROM 1.2?
3. There is an RTL bug may result in mailbox uncorrectable ECC error [issue](https://github.com/chipsalliance/caliptra-rtl/issues/340), is this fixed in ROM 1.2?

We plan to reference this in our internal integration work, so any information you can provide would be greatly appreciated.

Thanks!

mhatrevi, 2025-05-22:
@nquarton Can you address this please?

nquarton, 2025-06-03:
I would suggest reviewing the changes using the method above to determine if you would consider any of them to be critical issues for your particular usage model. However, I would not consider any of them to be critical from a security standpoint nor is it required for use with later 1.2.x RT FW.

Neal-liu, 2025-06-12:
Is ROM version 1.1.0 FIPS-certified? If we adopt Caliptra ROM 1.1.0, would that affect the SoC vendor’s ability to obtain FIPS certification?

Neal-liu, 2025-07-17:
Got it, thanks for your explanation.

closed 2025-07-17 by Neal-liu