chipsalliance/caliptra-sw #2183: Where and How owner PK Hash being defined

RaunakGu, 2025-05-21:
It seems Owner PK hash doesn't exist in the fuse map. It is used for verification as well as pcr extend?

1. Is there a plan to remove, if not where that is going to be defined?

jhand2, 2025-05-21:
Are you asking for 1.x or 2.x?

For 1.x, [fuse_owner_pk_hash](https://chipsalliance.github.io/caliptra-rtl/v1_1/external-regs/?p=caliptra_top_reg.generic_and_fuse_reg.fuse_owner_pk_hash%5B0%5D) is used.

For 2.x, it is moved to a register (instead of a fuse): https://chipsalliance.github.io/caliptra-rtl/main/external-regs/?p=caliptra_top_reg.generic_and_fuse_reg.CPTRA_OWNER_PK_HASH%5B0%5D

In both cases, the SoC can choose to read it from fuses or choose to populate it in some other way (e.g. getting it from a platform RoT, as defined in OCP ownership transfer spec).

RaunakGu, 2025-05-22:
My Question is for 2.x,

> the SoC can choose to read it from fuses.

- Do you mean SoC need to store to **its own** fuse and read and populate it?
- or Populate it by getting it from PaRoT.?

JohnTraverAmd, 2025-06-04:
Owner PK Hash was moved to a register just indicating that it doesn't need to be populated at fuse load time. The integrator can still chose to make this a fuse is they want.

Ultimately, if the integrator wants this enforced, the register must be populated before initializing the Load FW Command.  Where it comes from can be an integrator decision based on the security profile of the device.

RaunakGu, 2025-06-12:
https://github.com/chipsalliance/caliptra-ss/blob/main/src/fuse_ctrl/doc/otp_ctrl_field_descriptions.md
https://github.com/chipsalliance/caliptra-ss/pull/483

@jhand2, @JohnTraverAmd 
it seems “CPTRA_SS_OWNER_PK_HASH” got added back in the “VENDOR_HASHES_PROD_PARTITION” ?

ymo334, 2025-06-19:
@jhand2, @JohnTraverAmd, @RaunakGu 
I have a question about this.
There is only one CPTRA_SS_OWNER_PK_HASH field in efuse.
But considering Ownership Transfer, shouldn't there be more?

rstrongMCT, 2025-07-10:
there is a reference to the OCP Ownership Transfer specification. Assume that this is the spec: https://docs.google.com/document/d/1oANhjvv_R7E5n8w1RroN8l8-0jdYlfdQDp_3RqGV66k/edit#

If this is correct, this is not a published a published specification and is assumed that this is understood?