chipsalliance/caliptra-sw #2198: Caliptra Core 2.0 Mutable FW Checklist (FMC+RT)

jhand2, 2025-06-02:
This issue tracks features planned for the Caliptra 2.0 Firmware Release.

_Note: Features for 2.1 will be tracked separately_

* [X] All features in Caliptra 1.x (unless otherwise mentioned as "removed")
* [ ] X.509 ML-DSA DICE 
* [ ] PCR Signing (ECDSA P-384 & ML-DSA)
* [ ] Cryptographic Mailbox (CM)
    * [ ] AES (CBC, CTR, GCM)
    * [ ] ECDSA (non-DICE)
    * [ ] ECDH
    * [ ] HMAC / HKDF
    * [ ] ML-DSA (non-DICE)
    * [ ] SHA
* [ ] MCU Firmware Loading (Subsystem mode only)
* [ ] Update ECDSA & LMS very commands to pass hash instead of using sha512acc peripheral.
* [ ] Manifest authorization ML-DSA support

zhalvorsen, 2025-11-07:
Removing "Update invoke DPE command interface to support both ECC and ML-DSA profiles so there isn't a breaking change with 2.1 FW" because we are going to create a separate ML-DSA invoke_dpe command so it won't be an issue

zhalvorsen, 2025-11-12:
Removed "Consider increasing DPE context limit" because we won't have time to address it. We will need to treat this more similar to 1.x in #2685 because it will need to change the persistent data content in the future.

closed 2025-11-25 by mhatrevi