chipsalliance/caliptra-sw #2201: [Caliptra 1.2] No SVN field defined in manifest — how should manifest revocation be handled?

Neal-liu, 2025-06-03:
Hi team,

I’ve noticed that the Caliptra 1.2 manifest format (as described in [auth-manifest README](https://github.com/chipsalliance/caliptra-sw/blob/rt-1.2.0/auth-manifest/README.md)) doesn’t define any field for SVN (Security Version Number).

Given that, I’m wondering:
1. How should manifest revocation be implemented without a manifest SVN?
2. Is there any recommended workaround or planned support for this?
3. I understand that hardware SVN can be implemented using reserved SoC OTP fields. However, for software SVN (e.g., revoking older firmware images), it seems we would need to add an explicit SVN field into the manifest and have it covered by the signature — is that the intended direction?

Any guidance would be appreciated. Thanks!

mhatrevi, 2025-06-05:
SVN-based revocation is not supported for the auth manifest in version 1.2. The only way to revoke it is by replacing it with a new manifest.

Neal-liu, 2025-06-10:
Will version 2.0 be supported SVN-based revocation?
I do see SVN field in Auth Manifest, but I do not see SVN in fuse map.
Thanks.