chipsalliance/caliptra-sw #2202: Hitless Update - SoC Measurement stashed using AUTHORIZE_AND_STASH or STASH_MEASUREMENT causes context limit to exhaust

RaunakGu, 2025-06-03:
Hi @jhand2,

During Hitless update SoC measurements needed to be stashed using any of the command (AUTHORIZE_AND_STASH or STASH_MEASUREMENT).
This derives the context, maximum context allowed per Pauser is 16, 
So SoC can't exceed more than 16 stash, 

Caliptra RT FW by default stash MCU measurement during boot, so technically there are only 15 left.
[image]

How SoC supposed to handle this , or AM I missing something?

sunguk-bin-fad, 2025-06-10:
#2075 may help. While I believe there are areas for improvement, I think that the available options at this point are as follows.

- For stashing measurement, hitless updatable FWs should be measured with DeriveContext rather than STASH_MEASUREMENT. DeriveContext supports the RECURSIVE flag, and it will update the cumulative measurement and will not create new context.
- AUTHORIZE_AND_STASH supports the SKIP_STASH flag. 

To add one more point, the specification states that the Manifest supports up to 127 MetaData entries, but there is a technical limitation on the number that can be stashed.

@jhand2, Can you share your thoughts?

parvathib, 2026-10-08:
MCU recursively updates the cold-boot created SoC DPE contexts on hitless updates. No new contexts are created.

closed 2026-10-08 by parvathib