chipsalliance/caliptra-sw #2227: 2.0 - Encrypted FW Support for Boot(Streaming)

RaunakGu, 2025-06-05:
hi @mhatrevi ,

**Topic:** Caliptra supporting Encrypted Firmware for boot and update:

[image]

During Streaming boot, Caliptra FW(FMC,RT etc)  streamed using recovery interface to Caliptra, but Caliptra ROM/RT doesn't seems to have mechanism for handling encrypted images.
How do we handle this?

rstrongMCT, 2025-06-11:
Similar issue filed under: https://github.com/chipsalliance/Caliptra/issues/257

Closure of this as soon as possible is requested as this has SoC design/arch impact

mhatrevi, 2025-06-12:
Caliptra firmware components like FMC and RT are open source, and as such, the encryption requirement does not apply to them.

RaunakGu, 2025-06-12:
hi Vishal,

Caliptra FW(FMC/RT) is open source and encryption requirement doesn't apply to them,

But SoC Manifest and MCU RT is vendor specific firmware, for streaming boot these will be loaded by Caliptra RT firmware from recovery interface, verified and consumed,

So, technically we can't decrypt for MCU RT and SoC Manifest as well?

rstrongMCT, 2025-08-07:
In [caliptra-sw/runtime/src/recovery_flow.rs at main-2.x · chipsalliance/caliptra-sw](https://github.com/chipsalliance/caliptra-sw/blob/main-2.x/runtime/src/recovery_flow.rs) it appears that adding code to the following:

let digest = {
            let dma = &drivers.dma;
            let dma_recovery = DmaRecovery::new(
                drivers.soc_ifc.recovery_interface_base_addr().into(),
                drivers.soc_ifc.caliptra_base_axi_addr().into(),
                drivers.soc_ifc.mci_base_addr().into(),
                dma,
            );
            // need to make sure the device status is correct to load the next image
            dma_recovery.set_device_status(
                DmaRecovery::DEVICE_STATUS_READY_TO_ACCEPT_RECOVERY_IMAGE_VALUE,
            )?;
            cprintln!("[rt] Uploading MCU firmware");
            let mcu_size_bytes = dma_recovery.download_image_to_mcu(MCU_FIRMWARE_INDEX)?;

	// decrypt the downloaded image 

............. <insert code here>

            cprintln!("[rt] Calculating MCU digest");
            dma_recovery.sha384_mcu_sram(&mut drivers.sha2_512_384_acc, mcu_size_bytes)?
        };

The SoC Manifest requires modification since the decrypt operation will require entries additional data [optional] (aad), and the tag.  The associated decryption key (CKM) could be provided by MCU ROM to Caliptra early in boot as Caliptra will need this to decrypt the MCU RT FW. 

To consider: place the FW decrypt burden on BMC or equivalent - BMC decrypts the MCU and SoC FW components prior to streaming to Caliptra and MCU. This would technically meet the OCP 2.x requirements.