chipsalliance/caliptra-sw #2265: hitless update - Update reset processing

RaunakGu, 2025-06-20:
hi @mhatrevi

Few question for update reset processing w.r.t. Caliptra ROM/RT
 
In the hitless update flow:

1. The MCU RT issues a FIRMWARE_LOAD command.
2. The Caliptra RT asserts a fw_update reset, causing the Caliptra Core to reset and the flow to jump to the Caliptra ROM, initiating the update_reset sequence.
3. The Caliptra ROM streams the firmware from the mailbox and performs all necessary verifications.
4. If the Caliptra firmware verification is successful, the firmware is loaded, and the sequence proceeds with the core starting to run the FMC, followed by the RT.

Ques1: **If the Caliptra firmware verification fails, the execution needs to restore from the Caliptra ROM back to the Caliptra RT, Is it being done that way, or how does the invalid firmware verification case is handled.?**

Ques2: **During Update, FMC seems to be verified, and measurement is being captured in corresponding PCRs(PCR0, 1), But FMC doesn't gets loaded, so it happens that system still booted with old fmc copy. Is this expected?**

>         // Throw away the FMC portion of the image
>         txn.drop_words(manifest.fmc.size as usize / 4)?;

[image]

mhatrevi, 2025-06-26:
Ques1: If the Caliptra firmware verification fails, the execution needs to restore from the Caliptra ROM back to the Caliptra RT, Is it being done that way, or how does the invalid firmware verification case is handled.?
If the verification fails on an update reset, the RT is not updated in ICCM. Update reset is considered a non-fatal error and Caliptra ROM will continue to transfer control to the FMC and then from the FMC to the old RT.

Ques2: During Update, FMC seems to be verified, and measurement is being captured in corresponding PCRs(PCR0, 1), But FMC doesn't gets loaded, so it happens that system still booted with old fmc copy. Is this expected?
Yes, this is expected. FMC it not updated during an update reset, only on a cold reset.

RaunakGu, 2025-06-26:
hi Vishal, Thanks for the reply,

**Q1:** So, Caliptra will continue with Update reset flow?
This will have impact on Journey Measurements(PCR1, PCR3) and so DPE.? Measurements are old but it gets extended anyway for update reset.
PCR3(RT Journey Measurement) used for updating the dpe contexr(pls refer - update_dpe_rt_journey), how do we handle those?

**Q2:** If FMC is not updated and not running then shouldn't we skip extending the journey measurements for fmc(PCR1)?

cc @jhand2

mhatrevi, 2025-07-10:
Assigning to @jhand2 for DPE related question(s).

zhalvorsen, 2025-07-11:
> So, Caliptra will continue with Update reset flow?

Yes, it will continue to FMC and on to the original RT firmware

> This will have impact on Journey Measurements(PCR1, PCR3) and so DPE.?

Yes, the only way this actually affects DPE is through the function you listed there. I see what you are pointing out. If `A` hitlessly updates and fails it boots `A` again. This effectively measures `A` into the journey measurement twice. It will be running the same firmware, but with different DPE keys. The measurements in the certificate will be the same, but the derived keys will be different. I think this is the intention so it can prove the entire story of what the chip has verified and booted since the last cold boot.

> how do we handle those?

Do you mean from the verifier's perspective?

> Can we add a negative image verification(bad image) integration test case?

This is a great idea, I'll create some GitHub issues so we can track this. Besides checking the firmware boots back into the original firmware what else would you like us to check in those tests?

> If FMC is not updated and not running then shouldn't we skip extending the measurements for fmc(PCR0, PCR1)?

It still does run FMC after booting from an update reset. The boot path would look something like this:

`ColdBoot -> FMC -> RT -> UpdateReset -> FMC -> RT`